pfalcon/esp-open-sdk #183: Important SDK Patch Released

fvpalha, 2016-07-12:
Hi @pfalcon 

Do you have a schedule to add the [Important SDK Patch Released](http://bbs.espressif.com/viewtopic.php?f=46&t=2407)?

Thank you.

fvpalha, 2016-07-21:
Hi @pfalcon 

Now they launched [ESP8266_NONOS_SDK_V2.0.0_16_07_19](http://bbs.espressif.com/viewtopic.php?f=46&t=2451).

Thank you.

jeremyd2019, 2016-07-27:
I made patches for the patch for 1.5.4, and to add the sdk 2.0.0.

I have not tested compiling anything with either yet, but hopefully having a patch/pull request out there will get more eyes on it than just mine.

fvpalha, 2016-07-27:
Thank you @jeremyd2019

stevemk14ebr, 2016-08-10:
Attempt to build toolchain with SDK 2.0.0

> 2016-08-09 21:36:41 (6.30 KB/s) - ‘ESP8266_NONOS_SDK_V2.0.0_16_07_19.zip’ saved [5122726/5122726]
> 
> unzip -q -o ESP8266_NONOS_SDK_V2.0.0_16_07_19.zip
> mv ESP8266_NONOS_SDK ESP8266_NONOS_SDK_V2.0.0_16_07_19
> mv License ESP8266_NONOS_SDK_V2.0.0_16_07_19
> mv: cannot stat ‘License’: No such file or directory
> make: [ESP8266_NONOS_SDK_V2.0.0_16_07_19/.dir] Error 1 (ignored)
> touch ESP8266_NONOS_SDK_V2.0.0_16_07_19/.dir
> ln -snf ESP8266_NONOS_SDK_V2.0.0_16_07_19 sdk
> /home/steve/ESPSDK2.0/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc -O2 -c user_rf_cal_sector_set.c
> user_rf_cal_sector_set.c:1:21: fatal error: c_types.h: No such file or directory
>  #include <c_types.h>
>                      ^
> compilation terminated.
> make: *** [user_rf_cal_sector_set.o] Error 1
> steve@steve-VirtualBox:~/ESPSDK2.0/esp-open-sdk$ 
> 

Solutions?

jeremyd2019, 2016-08-10:
@stevemk14ebr - see PR #187

stevemk14ebr, 2016-08-10:
I'm attempting those modifications now. FYI they released a patch to 2.0.0 today:
http://bbs.espressif.com/viewtopic.php?f=46&t=2529

jrobeson, 2016-10-19:
Is this still a problem? master uses SDK 2.0 now.

pfalcon, 2016-10-19:
Closing per above.

closed 2016-10-19 by pfalcon