pfalcon/esp-open-sdk #194: Build fails on Linux Mint

andrepanno, 2016-08-20:
I followed all the installation procedures correctly, but for some reason when I try to build the  `esp-open-sdk` I receive the following errors: 
`blackrad@blackrad-sys ~/Sviluppo/esp8266/esp-open-sdk $ make
Makefile:180: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V1.5.4_16_05_20/.dir'
Makefile:169: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V1.5.4_16_05_20/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory '/home/blackrad/Sviluppo/esp8266/esp-open-sdk/crosstool-NG'
../Makefile:180: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V1.5.4_16_05_20/.dir'
../Makefile:169: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V1.5.4_16_05_20/.dir'
./bootstrap
Running autoconf...
./bootstrap: 5: ./bootstrap: autoconf: not found
../Makefile:136: recipe for target '_ct-ng' failed
make[1]: *** [_ct-ng] Error 127
make[1]: Leaving directory '/home/blackrad/Sviluppo/esp8266/esp-open-sdk/crosstool-NG'
Makefile:133: recipe for target 'crosstool-NG/ct-ng' failed
make: *** [crosstool-NG/ct-ng] Error 2`

Does anyone have an idea to resolve this problem?

yenthusiastic, 2016-08-26:
i have the same error when trying to build on Raspberry Pi Raspian Jessie. Hoping to find a solution soon!

heinervdm, 2016-08-26:
`./bootstrap: autoconf: not found`
As this line tells you, you have no autoconf installed, just install it. Normally it should be part of build-essentials...

andrepanno, 2016-08-26:
@heinervdm I'll give it a try as soon as possible!

yenthusiastic, 2016-08-30:
Thanks! I've tried it and it worked! The problem is I did not install the whole build-essentials package so had to do that before "make". Also, I realised it is necessary to "sudo apt-get update" before installing all the build-essentials to prevent errors from constantly appearing.

andrepanno, 2016-08-31:
@yenthusiastic exactly 👍

jrobeson, 2016-10-19:
`autoconf` is included in the README. Is there anything else that should be done as part of this ticket?

irwanayeah, 2019-05-20:
using linux mint Linux Mint 19.1

> 
> Makefile:186: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
> Makefile:181: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
> make -C crosstool-NG -f ../Makefile _ct-ng
> make[1]: Entering directory '/home/ayeah/esp-open-sdk/crosstool-NG'
> ../Makefile:186: warning: overriding recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
> ../Makefile:181: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
> ./bootstrap
> Running autoconf...
> Done. You may now run:
>     ./configure
> ./configure --prefix=`pwd`
> checking build system type... x86_64-pc-linux-gnu
> checking host system type... x86_64-pc-linux-gnu
> checking for a BSD-compatible install... /usr/bin/install -c
> checking for grep that handles long lines and -e... /bin/grep
> checking for egrep... /bin/grep -E
> checking for a sed that does not truncate output... /bin/sed
> checking whether sed understands -r -i -e... yes
> checking whether ln -s works... yes
> checking for gcc... gcc
> checking whether the C compiler works... yes
> checking for C compiler default output file name... a.out
> checking for suffix of executables... 
> checking whether we are cross compiling... no
> checking for suffix of object files... o
> checking whether we are using the GNU C compiler... yes
> checking whether gcc accepts -g... yes
> checking for gcc option to accept ISO C89... none needed
> checking how to run the C preprocessor... gcc -E
> checking for ranlib... ranlib
> checking for gobjcopy... no
> checking for objcopy... objcopy
> checking for absolute path to objcopy... /usr/bin/objcopy
> checking for gobjdump... no
> checking for objdump... objdump
> checking for absolute path to objdump... /usr/bin/objdump
> checking for greadelf... no
> checking for readelf... readelf
> checking for absolute path to readelf... /usr/bin/readelf
> checking for gperf... gperf
> checking for absolute path to gperf... /usr/bin/gperf
> checking for bison... bison
> checking for flex... flex
> checking for makeinfo... makeinfo
> checking for cut... cut
> checking for stat... stat
> checking for readlink... readlink
> checking for wget... wget
> checking for tar... tar
> checking for gzip... gzip
> checking for bzip2... bzip2
> checking for help2man... help2man
> checking for gpatch... no
> checking for patch... patch
> checking for absolute path to patch... /usr/bin/patch
> checking for bash >= 3.1... /bin/bash
> checking for GNU awk... /usr/bin/gawk
> checking for GNU make >= 3.80... /usr/bin/make
> checking whether /usr/bin/make sets $(MAKE)... yes
> checking for make 3.81... no
> checking for GNU libtool >= 1.5.26... /usr/bin/libtool
> checking for GNU libtoolize >= 1.5.26... /usr/bin/libtoolize
> checking for GNU automake >= 1.10... /usr/bin/automake
> checking for the type used in gperf declarations... size_t
> checking for xz... xz
> checking for cvs... no
> checking for svn... no
[... 142 lines cut ...]
> [ERROR]  >>
> [ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
> [ERROR]  >>        called in step '(top-level)'
> [ERROR]  >>
> [ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
> [ERROR]  >>        called from: CT_Extract[scripts/functions@982]
> [ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
> [ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
> [ERROR]  >>
> [ERROR]  >>  For more info on this error, look at the file: 'build.log'
> [ERROR]  >>  There is a list of known issues, some with workarounds, in:
> [ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
> [ERROR]   
> [ERROR]  (elapsed: 0:02.85)
> [00:03] / ct-ng:152: recipe for target 'build' failed
> make[2]: *** [build] Error 2
> make[2]: Leaving directory '/home/ayeah/esp-open-sdk/crosstool-NG'
> ../Makefile:135: recipe for target '_toolchain' failed
> make[1]: *** [_toolchain] Error 2
> make[1]: Leaving directory '/home/ayeah/esp-open-sdk/crosstool-NG'
> Makefile:130: recipe for target 'crosstool-NG/.built' failed
> make: *** [crosstool-NG/.built] Error 2
> 

how to fix them?

OladapoAjala, 2019-08-09:
@irwanayeah Thanks Awesome stuff out there....