pfalcon/esp-open-sdk #198: Merge lwip patch from nodemcu?

cmarrin, 2016-09-12:
The Nodemcu repo, which also has lwip source, has made a patch to resolve an mdns issue I'm having. The patch is here:

    https://github.com/nodemcu/nodemcu-firmware/pull/1192

I'm wondering how I should proceed. If your SDK has the right patches it would be great to just use it directly. Or I could fork your SDK and patch it locally. Or I could patch the lwip source in the 2.0 SDK from Espressif and use that.

If you want the patch, let me know and I'll fork and submit the pull request

pfalcon, 2016-09-13:
The first step is figuring out whether that "mdns" code is actually from lwIP or not. If not, it should be submitted upstream, accepted to it, then ESP8266 port upgraded to the latest lwIP. If mdns code is already from upstream, the above should be done for changes in https://github.com/nodemcu/nodemcu-firmware/pull/1192 (full upgrade of esp8266 port is still required of course). Feel free to start with any of these.

cmarrin, 2016-09-13:
Here's my current plan. I've forked pfalcon/esp-open-lwip. I'm building it with Makefile.open and now I'm in the process of switching my code to use lwip directly instead of using the espconn wrappers. Once that's done I'll look at the nodemcu patch(es) and/or debug the mdns problems I'm seeing and then figure out where it should go.

How close to the upstream lwip code is your version?

pfalcon, 2017-05-21:
See https://github.com/pfalcon/esp-open-sdk/issues/263

closed 2017-05-21 by pfalcon