pfalcon/esp-open-sdk #199: Compilation fails with a couple of files not found.

jitsuslap, 2016-09-13:
I have made the following makefile:

```export PATH=/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:$PATH
LIB=-L/Volumes/case-sensitive/esp-open-sdk/sdk/lib
INC=-I/Volumes/case-sensitive/esp-open-sdk/sdk/include

all:
	echo $(PATH)
	xtensa-lx106-elf-gcc $(LIB) $(INC) *.c -o output
```
I get the following;
*echo /Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:ATH
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin:ATH
xtensa-lx106-elf-gcc -L/Volumes/case-sensitive/esp-open-sdk/sdk/lib -I/Volumes/case-sensitive/esp-open-sdk/sdk/include *.c -o output
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find crt1-sim.o: No such file or directory
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find _vectors.o: No such file or directory
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -lsim
/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -lhandlers-sim
collect2: error: ld returned 1 exit status
make: *** [all] Error 1*

I need help.

jcmvbkbc, 2016-09-13:
There are actually no such files there. This way of compiler invocation is meant to produce a complete executable, but the compiler is not configured to do that. Although it seems possible to write a spec that will default to the esp8266-specific libraries and startup files.

jitsuslap, 2016-09-15:
Okay .... That's no help at all. Why should this happen at all?? Why isn't it all configured properly already?? I don't have the know-how to go tinkering down the hood.
      Anybody else wanna chime in and help. Does anyone know an alternative to this git?
Thanks for the input anyways @jcmvbkbc jcmvbkbc. Appreciate that.

cranphin, 2016-09-15:
You might want to lead with what you are trying to achieve, I assume you just want to get some simple code to run on a ESP8266? If so I'd recommend using the Arduino IDE + the Arduino ESP8266 core, it's a lot easier to use then esp-open-sdk. Alternatively look at the examples folder for a sample, but expect frustration without proper background knowledge :) Keep in mind esp-open-sdk is at it's base just a Makefile to pull together and compile all the parts needed to form a workable SDK, biased towards unix environments, and nothing more then that, it's not made for ease of use :)

jcmvbkbc, 2016-09-15:
> I don't have the know-how to go tinkering down the hood.

Yes, sorry, I should have imagined that. And my reply was more of a statement that what you see is expected and a reminder to myself of what could be improved. I'll definitely look at how to make it work.

In the meantime please use example projects circulating around the community, like https://github.com/pfalcon/esp-open-sdk/tree/master/examples/blinky or ones at http://git.spritesserver.nl/ or https://github.com/SuperHouse/esp-open-rtos

jitsuslap, 2016-09-16:
@jcmvbkbc Thanks for the links! 
@cranphin I wanted to write my code for NodeMCU in C, like I do for Arduino. I like the lean and mean code that makes, but I think it's more of an OCD thing. For some reason the Arduino IDE is not compiling the code in pure C but only the code in "Arduino type C". I can upload the code just fine. Thanks for the input anyways. I am a total noob yet.

closed 2016-10-19 by pfalcon

jcmvbkbc, 2016-10-19:
Hello,

as an attempt to fix the original issue and improve new user experience [this branch](https://github.com/pfalcon/esp-open-sdk/compare/master...jcmvbkbc:esp8266-spec?expand=1) adds default spec to the compiler that allows building simple esp8266 projects with mere ```xtensa-lx106-elf-gcc source.c -o project.elf```.

drawkula, 2016-10-19:
@jitsuslap: Exports in GNU-`make` look like:

    export PATH := /opt/esp-open-sdk/xtensa-lx106-elf/bin:$(PATH)

jitsuslap, 2016-11-02:
Thanks a lot man. You rock! 
I had learnt making a complete makefile though but I imagine this will be really helpful to newbies.