pfalcon/esp-open-sdk #200: Build CRASHES Raspberry Pi A+

unorthodox-engineers, 2016-09-16:
I've tried this three times now, and every time I build the library, it gets up to the "stage 2 GCC install" of the cross-compiler, and hard crashes the machine.  As in the network goes down, the error light comes on, and the machine doesn't respond until it's rebooted. 

Always happens in the same place, after FIVE HOURS of build. It's getting pretty frustrating. This is day three.

And before you ask why I'm trying to set it up on a Pi A+ / Zero instead of a "real" linux computer, it's is a portable $5 machine with 3.3v level UART and GPIO capable of compiling C code and directly interfacing with the ESP. Seems obvious to me. 

It works for all my other embedded development, so It's a real shame your SDK won't compile. I'll have to keep using the Arduino IDE on my primary windows machine.

jeremyd2019, 2016-09-16:
It built for me on my Raspberry Pi 2 Model B, and on my Fire 7" tablet (the one that was on sale for $30 at Black Friday) running a debian chroot.

udif, 2016-09-16:
According to Wikipedia, most A+ boards shipped with 256MB RAM, and only fairly recent A+ boards (May 2016 onwards) have increased the memory size to 512MB. Perhaps your board doesn't have enough memory for this task? 
You could also check how your memory is split between the GPU and CPU, and perhaps increase the CPU memory at the expense of the GPU.

pfalcon, 2017-05-21:
See #101.

closed 2017-05-21 by pfalcon