pfalcon/esp-open-sdk #204: How do I build stuff?

jvikstrom, 2016-10-07:
Hi, I'm trying to compile a program using this sdk. It does not work though.
Whenever I try to run "gen_misc.sh" and after having entered all the values it fails with this error message: "make: Nothing to be done for 'FORCE'.".
If I try to issue the "make" command I also get that same error.

I copied once from "sdk/examples/lwip_open_src_template_proj", once from here: " https://github.com/pfalcon/esp-open-lwip/tree/8c39d2179a273553466043f388772abb6251a4ca ", and none of them work.
I also tried to compile the examples under sdk. And I still get that same error.

HOWEVER. I can compile the blinky example that is in this repo ("examples/blinky") when issuing the "make" command.

I've tried to reinstall the sdk. (Doing make clean...)

So what am I doing wrong?

Thank you all in advance!

cranphin, 2016-10-07:
I don't think the samples in LWIP (Lightweight IP) are meant to be built as examples in esp-open-sdk. Although to be honest I'm not sure why they are there.

The blinky sample -is- meant as an example to be built using esp-open-sdk, and since you managed to do this, I'd say you succeeded :)

It's the -only- example for reasons mentioned in the readme in the example folder:
"The directory would contain complete source for examples, but so far
examples with suitable OpenSource licensing were not found."

So if you're looking for other examples, you will have to find them on github. Preferably ones that mention they use esp-open-sdk, and how to build them.

Or create your own by expanding the blinky example :)

jvikstrom, 2016-10-07:
Aha, I see. Guess I'll just steal the blinky makefile then.
I got quite confused by the sdk examples tbh, but now I know how to progress. 

So thank you for making things clearer :p

closed 2016-10-07 by jvikstrom

pfalcon, 2016-10-07:
I've pushed https://github.com/pfalcon/esp-open-sdk/commit/2d7d17069852031b04f623ff43d7f46ef349baf0 to clarify these matters.