pfalcon/esp-open-sdk #205: Problem with installing esp-open-sdk on ubuntu 16.04

MustafaFayez, 2016-10-10:
I tried to install esp-open-sdk on ubuntu 16.04 
when i called make STANDALONE=y  it gives errors in building the repo
I removed 16.04 and returned back to 14.04 and it worked just fine.
Plz consider 16.04 compatibility.
@pfalcon

jcmvbkbc, 2016-10-10:
> when i called make STANDALONE=y it gives errors in building the repo

Care to post error messages/relevant build logs?

MustafaFayez, 2016-10-10:
make: Entering directory `/opt/esp-open-sdk/crosstool-NG'
  SED    'ct-ng'
  SED    'scripts/crosstool-NG.sh'
  SED    'scripts/saveSample.sh'
  SED    'scripts/showTuple.sh'
  GEN    'config/configure.in'
  GEN    'paths.mk'
  GEN    'paths.sh'
  DEP    'nconf.gui.dep'
  DEP    'nconf.dep'
  DEP    'lxdialog/yesno.dep'
  DEP    'lxdialog/util.dep'
  DEP    'lxdialog/textbox.dep'
  DEP    'lxdialog/menubox.dep'
  DEP    'lxdialog/inputbox.dep'
  DEP    'lxdialog/checklist.dep'
  DEP    'mconf.dep'
  DEP    'conf.dep'
  BISON  'zconf.tab.c'
  GPERF  'zconf.hash.c'
  LEX    'lex.zconf.c'
  DEP    'zconf.tab.dep'
  CC     'zconf.tab.o'
zconf.tab.c: In function `zconfparse':
zconf.tab.c:1656: error: `kconf_id_strings' undeclared (first use in this functi
on)
zconf.tab.c:1656: error: (Each undeclared identifier is reported only once
zconf.tab.c:1656: error: for each function it appears in.)
zconf.tab.c:1764: warning: initialization makes pointer from integer without a c
ast
zconf.tab.c: In function `zconf_endtoken':
zconf.tab.c:2315: error: `kconf_id_strings' undeclared (first use in this functi
on)
In file included from zconf.tab.c:2507:
lex.zconf.c: In function `zconflex':
lex.zconf.c:1103: warning: initialization makes pointer from integer without a c
ast
lex.zconf.c:1178: warning: initialization makes pointer from integer without a c
ast
make: *** [zconf.tab.o] Error 1
make: *** [build-lib-kconfig] Error 2
make: *** [build] Error 2
make: Leaving directory `/opt/esp-open-sdk/crosstool-NG'
make[1]: *** [_ct-ng] Error 2
make[1]: Leaving directory `/optes
p-open-sdk/crosstool-NG'
make: *** [crosstool-NG/ct-ng] Error 2

edtler, 2016-10-16:
Hi, 
I've solved thi problem. 
Not all dependencies are installed you have to launch all dependencies with the next command:

sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf \
    flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
    sed git unzip bash help2man wget bzip2  libtool-bin

It's the same for Ubuntu 14.04

closed 2016-10-19 by pfalcon