pfalcon/esp-open-sdk #212: Cygwin compilation error - collect2: error: ld returned 1 exit status

ArchontPL, 2016-10-23:
Hello,
I want to compile SDK under Windows 8.1 using Cygwin
```
$ uname -a
CYGWIN_NT-6.3 T420s 2.6.0(0.304/5/3) 2016-08-31 14:32 x86_64 Cygwin
```
I use `make STANDALONE=n` and I still get `collect2: error: ld returned 1 exit status` error
Last lines from build.log

```
[ALL  ]    Making init.c
[ALL  ]    x86_64-build_unknown-cygwin-gcc -g -O2   -I. -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/common -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/config -DLOCALEDIR=""/home/Archont/ESP2866/esp-open-sdk/xtensa-lx106-elf/share/locale"" -DHAVE_CONFIG_H -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../include/opcode -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../opcodes/.. -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../readline/.. -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../zlib -I../bfd -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../bfd -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../include -I../libdecnumber -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/../libdecnumber  -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/gnulib/import -Ibuild-gnulib/import   -DTUI=1  -I/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  -I/usr/include/python2.7 -I/usr/include/python2.7 -Wall -Wpointer-arith -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wempty-body -Wpointer-sign -Wmissing-prototypes -Wdeclaration-after-statement -Wmissing-parameter-type -Wold-style-declaration -Wold-style-definition -Wformat-nonliteral  -c -o init.o -MT init.o -MMD -MP -MF .deps/init.Tpo init.c
[ALL  ]    rm -f gdb.exe
[ALL  ]    x86_64-build_unknown-cygwin-gcc -g -O2   -static-libstdc++ -static-libgcc -Wl,--stack,12582912   	-o gdb.exe gdb.o xtensa-tdep.o xtensa-config.o solib-svr4.o ser-base.o ser-unix.o ser-pipe.o ser-tcp.o  remote.o dcache.o tracepoint.o ax-general.o ax-gdb.o remote-fileio.o remote-notif.o ctf.o tracefile.o tracefile-tfile.o  cli-dump.o cli-decode.o cli-script.o cli-cmds.o cli-setshow.o cli-logging.o cli-interp.o cli-utils.o mi-out.o mi-console.o mi-cmds.o mi-cmd-catch.o mi-cmd-env.o mi-cmd-var.o mi-cmd-break.o mi-cmd-stack.o mi-cmd-file.o mi-cmd-disas.o mi-symbol-cmds.o mi-cmd-target.o mi-cmd-info.o mi-interp.o mi-main.o mi-parse.o mi-getopt.o tui-command.o tui-data.o tui-disasm.o tui-file.o tui-hooks.o tui-interp.o tui-io.o tui-layout.o tui-out.o tui-regs.o tui-source.o tui-stack.o tui-win.o tui-windata.o tui-wingeneral.o tui-winsource.o tui.o python.o py-arch.o py-auto-load.o py-block.o py-bpevent.o py-breakpoint.o py-cmd.o py-continueevent.o py-xmethods.o py-event.o py-evtregistry.o py-evts.o py-exitedevent.o py-finishbreakpoint.o py-frame.o py-framefilter.o py-function.o py-gdb-readline.o py-inferior.o py-infevents.o py-infthread.o py-lazy-string.o py-linetable.o py-newobjfileevent.o py-objfile.o py-param.o py-prettyprint.o py-progspace.o py-signalevent.o py-stopevent.o py-symbol.o py-symtab.o py-threadevent.o py-type.o py-unwind.o py-utils.o py-value.o py-varobj.o guile.o elfread.o stap-probe.o dtrace-probe.o posix-hdep.o posix-strerror.o c-exp.o cp-name-parser.o ada-exp.o jv-exp.o d-exp.o f-exp.o go-exp.o m2-exp.o p-exp.o version.o annotate.o addrmap.o auto-load.o auxv.o agent.o bfd-target.o blockframe.o breakpoint.o break-catch-sig.o break-catch-throw.o break-catch-syscall.o findvar.o regcache.o cleanups.o charset.o continuations.o corelow.o disasm.o dummy-frame.o dfp.o source.o value.o eval.o valops.o valarith.o valprint.o printcmd.o block.o symtab.o psymtab.o symfile.o symfile-debug.o symmisc.o linespec.o dictionary.o infcall.o infcmd.o infrun.o expprint.o environ.o stack.o thread.o exceptions.o extension.o filesystem.o filestuff.o inf-child.o interps.o minidebug.o main.o macrotab.o macrocmd.o macroexp.o macroscope.o mi-common.o event-loop.o event-top.o inf-loop.o completer.o gdbarch.o arch-utils.o gdbtypes.o gdb_bfd.o gdb_obstack.o osabi.o copying.o memattr.o mem-break.o target.o target-dcache.o parse.o language.o build-id.o buildsym.o findcmd.o std-regs.o signals.o exec.o reverse.o bcache.o objfiles.o observer.o minsyms.o maint.o demangle.o dbxread.o coffread.o coff-pe-read.o dwarf2read.o mipsread.o stabsread.o corefile.o dwarf2expr.o dwarf2loc.o dwarf2-frame.o dwarf2-frame-tailcall.o ada-lang.o c-lang.o d-lang.o f-lang.o objc-lang.o ada-tasks.o ada-varobj.o c-varobj.o ui-out.o cli-out.o varobj.o vec.o go-lang.o go-valprint.o go-typeprint.o jv-lang.o jv-valprint.o jv-typeprint.o jv-varobj.o m2-lang.o opencl-lang.o p-lang.o p-typeprint.o p-valprint.o sentinel-frame.o complaints.o typeprint.o ada-typeprint.o c-typeprint.o f-typeprint.o m2-typeprint.o ada-valprint.o c-valprint.o cp-valprint.o d-valprint.o f-valprint.o m2-valprint.o serial.o mdebugread.o top.o utils.o ui-file.o user-regs.o frame.o frame-unwind.o doublest.o frame-base.o inline-frame.o gnu-v2-abi.o gnu-v3-abi.o cp-abi.o cp-support.o cp-namespace.o reggroups.o trad-frame.o tramp-frame.o solib.o solib-target.o prologue-value.o memory-map.o memrange.o xml-support.o xml-syscall.o xml-utils.o target-descriptions.o target-memory.o xml-tdesc.o xml-builtin.o inferior.o osdata.o gdb_usleep.o record.o record-full.o gcore.o gdb_vecs.o jit.o progspace.o skip.o probe.o common-utils.o buffer.o ptid.o gdb-dlfcn.o common-agent.o format.o registry.o btrace.o record-btrace.o waitstatus.o print-utils.o rsp-low.o errors.o common-debug.o debug.o common-exceptions.o btrace-common.o fileio.o compile.o compile-c-symbols.o compile-c-types.o compile-object-load.o compile-object-run.o compile-loc2c.o compile-c-support.o inflow.o    init.o 	   ../readline/libreadline.a ../opcodes/libopcodes.a ../bfd/libbfd.a -L./../zlib -lz  ../libiberty/libiberty.a ../libdecnumber/libdecnumber.a    -lncurses -lm     -ldl -lpython2.7 /home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/lib/libexpat.a    ../libiberty/libiberty.a -luser32 build-gnulib/import/libgnu.a 
[ALL  ]    tui-win.o:tui-win.c:(.rdata$.refptr.stdscr[.refptr.stdscr]+0x0): undefined reference to `stdscr'
[ALL  ]    tui-win.o:tui-win.c:(.rdata$.refptr.curscr[.refptr.curscr]+0x0): undefined reference to `curscr'
[ALL  ]    tui-win.o:tui-win.c:(.rdata$.refptr.acs_map[.refptr.acs_map]+0x0): undefined reference to `acs_map'
[ALL  ]    tui.o:tui.c:(.rdata$.refptr.COLS[.refptr.COLS]+0x0): undefined reference to `COLS'
[ALL  ]    tui.o:tui.c:(.rdata$.refptr.LINES[.refptr.LINES]+0x0): undefined reference to `LINES'
[ERROR]    collect2: error: ld returned 1 exit status
[ERROR]    make[5]: *** [Makefile:1391: gdb.exe] Error 1
[ALL  ]    make[5]: Leaving directory '/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-gdb-cross/gdb'
[ERROR]    make[4]: *** [Makefile:8759: all-gdb] Error 2
[ALL  ]    make[4]: Leaving directory '/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-gdb-cross'
[ERROR]    make[3]: *** [Makefile:849: all] Error 2
[ALL  ]    make[3]: Leaving directory '/home/Archont/ESP2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-gdb-cross'
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a-dirty/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 81:16.19)
```
Any ideas how to fix this?

cranphin, 2016-10-24:
This looks similar, mayby it helps? https://github.com/crosstool-ng/crosstool-ng/issues/321

ArchontPL, 2016-10-24:
OK, this was a good tip. I found also other issues (I will request pull when I get SDK running on Cygwin), but now I cannot compile lx106-hal.
```
$ make -C lx106-hal -f ../Makefile _libhal
make: enter directory '/home/Archont/esp2866/esp-open-sdk/lx106-hal'
../Makefile:185: warning: commands covering for object 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:180: warning: ignoring old command for object 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
autoreconf -i
Unescaped left brace in regex is deprecated, passed through in regex; marked by <-- HERE in m/\${ <-- HERE ([^ \t=:+{}]+)}/ at /usr/bin/automake-1.14 line 3931.
PATH="/home/Archont/esp2866/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/bin:/usr/bin:/cygdrive/c/ProgramData/Oracle/Java/javapath:/cygdrive/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/cygdrive/c/Windows/system32:/cygdrive/c/Windows:/cygdrive/c/Windows/System32/Wbem:/cygdrive/c/Windows/System32/WindowsPowerShell/v1.0:/cygdrive/c/Program Files (x86)/Intel/OpenCL SDK/2.0/bin/x86:/cygdrive/c/Program Files (x86)/Intel/OpenCL SDK/2.0/bin/x64:/cygdrive/c/Program Files (x86)/Calibre2:/cygdrive/c/Program Files/MATLAB/MATLAB Production Server/R2015a/runtime/win64:/cygdrive/c/Program Files/MATLAB/MATLAB Production Server/R2015a/bin:/cygdrive/c/Program Files/MATLAB/MATLAB Production Server/R2015a/polyspace/bin:/cygdrive/c/Program Files/KiCad/KicadBOM" ./configure --host=xtensa-lx106-elf --prefix=/home/Archont/esp2866/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for xtensa-lx106-elf-strip... xtensa-lx106-elf-strip
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking build system type... x86_64-unknown-cygwin
checking host system type... xtensa-lx106-elf
checking for style of include used by make... GNU
checking for xtensa-lx106-elf-gcc... xtensa-lx106-elf-gcc
checking whether the C compiler works... no
configure: error: in `/home/Archont/esp2866/esp-open-sdk/lx106-hal':
configure: error: C compiler cannot create executables
See `config.log' for more details
make: *** [../Makefile:167: _libhal] Error 77
make: leaving directory '/home/Archont/esp2866/esp-open-sdk/lx106-hal'
```
This can be connected with my 2 errors during xtensa-lx106-elf compilation? Because I got:
```
[ALL  ]    x86_64-build_unknown-cygwin-g++ -c   -O2 -g -pipe  -DIN_GCC -DCROSS_DIRECTORY_STRUCTURE  -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual    -DHAVE_CONFIG_H -I. -Ic-family -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/c-family -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/../include -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/../libcpp/include -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/../libdecnumber -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/../libbacktrace -DCLOOG_INT_GMP -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include -I/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/buildtools/include  /home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/src/gcc-4.8.5/gcc/c-family/cppspec.c -o c-family/cppspec.o
[ALL  ]    echo | /home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/xgcc -B/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/ -E -dM - |   sed -n -e 's/^#define ([^_][a-zA-Z0-9_]*).*/1/p' 	 -e 's/^#define (_[^_A-Z][a-zA-Z0-9_]*).*/1/p' |   sort -u > tmp-macro_list
[ALL  ]    if /home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/xgcc -B/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/ -print-sysroot-headers-suffix > /dev/null 2>&1; then   set -e; for ml in `/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/xgcc -B/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/ -print-multi-lib`; do     multi_dir=`echo ${ml} | sed -e 's/;.*$//'`;     flags=`echo ${ml} | sed -e 's/^[^;]*;//' -e 's/@/ -/g'`;     sfx=`/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/xgcc -B/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-final/./gcc/ ${flags} -print-sysroot-headers-suffix`;     if [ "${multi_dir}" = "." ];       then multi_dir="";     else       multi_dir=/${multi_dir};     fi;     echo "${sfx};${multi_dir}";   done; else   echo ";"; fi > tmp-fixinc_list
[ERROR]    xgcc: error: spawn: No such file or directory

...

[... cut ...] nt/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-core-pass-2/./gcc/xgcc -B/home/Archont/esp2866/esp-open-sdk/crosstool-NG/.build/xtensa-lx106-elf/build/build-cc-gcc-core-pass-2/./gcc/ ${flags} -print-sysroot-headers-suffix`;     if [ "${multi_dir}" = "." ];       then multi_dir="";     else       multi_dir=/${multi_dir};     fi;     echo "${sfx};${multi_dir}";   done; else   echo ";"; fi > tmp-fixinc_list
[ERROR]    xgcc: error: spawn: No such file or directory
```

davydnorris, 2016-10-25:
I had same issues.

For the first one, add --disable-tui to the gdb flags
For the second one, either use the patch I have supplied or, even better, switch to gcc5.2.0 and the problem will go away.

See my posts here: [http://www.esp8266.com/viewtopic.php?f=9&t=11782](url)

ArchontPL, 2016-10-25:
@davydnorris as I write before - I succesfully compile cross-gdb (also by adding -disable-tui). Now I'd like to compile libhal using your workaround, but alter 'make clean' file gcc/config.host is deleted, so I don't know how to recompile all SDK.

davydnorris, 2016-10-25:
So you have two options:
 - use my patch by saving it to a file in local-patches/gcc/4.8.5 where it will be automatically applied
 - use ct-ng menuconfig and change gcc to 5.2.0

I started with the first one but then changed to gcc5.2.0 (actually 5.4.0 now)

ArchontPL, 2016-10-25:
@davydnorris thanks! Now all project compile without problems. I use GCC 4.8.5 with yours patch, maybe in future crosstool-NG will switch to 5.4.0 version.
I added pull request with my findings to esp-open-sdk and crosstool-NG. I also pull request your patch for  crosstool-NG (of course with information about author) - I hope that you don't mind :).

closed 2016-10-25 by ArchontPL