pfalcon/esp-open-sdk #220: WPA2-PEAP doesn't work

gorghino, 2016-10-28:
I tried WPA2-PEAP support with your OpenSDK but something doesn't work.

**WiFi Module**: ESP8266-07
**user_main.c:** http://pastebin.com/4ER5cmN2
**Makefile**: http://pastebin.com/FR3wmkMC
**Output**:
d��d` ��s�l��
Connecting..
WPA2 ENTERPRISE VERSION: [v2.0] enable
mode : sta(18:fe:34:a5:84:10)
add if0
scandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
pm open,type:2 0
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
pm open,type:2 0
state: 5 -> 0 (2)
rm 0
pm close 7
reconnect
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt

(I removed "No IP found" from output)

wifi_station_get_connect_status() returns "STATION_WRONG_PASSWORD" but I tried with two working account (Tested on Ubuntu and Android).

jtuttas, 2017-01-20:
@gorghino !
Did you have found a solution for this problem, my logging looks exactly the same (the only difference is i have 'aid 47' instead of your 'aid 2') and i can'nt get a IP Adress from out DHCP Server as well?

best regards

jtuttas

gorghino, 2017-01-20:
Unfortunately not,
I moved to another board because of this bug :/
But I saw you've already found this workaround (https://github.com/esp8266/Arduino/issues/1032#issuecomment-273441710)