pfalcon/esp-open-sdk #221: CT_DoForceRmdir failure in Mac OS Sierra

jacobjennings, 2016-11-05:
[INFO ]  Installing final gcc compiler
[INFO ]  Installing final gcc compiler: done in 448.51s (at 30:41)
[INFO ]  =================================================================
[INFO ]  Installing cross-gdb
[INFO ]  Installing cross-gdb: done in 303.62s (at 35:45)
[INFO ]  =================================================================
[INFO ]  Cleaning-up the toolchain's directory
[INFO ]    Stripping all toolchain executables
**[35:48] / stat: cannot read file system information for '%Lp': No such file or directory**
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Cleaning-up the toolchain's directory'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoForceRmdir[scripts/functions@460]
[ERROR]  >>        called from: do_finish[scripts/build/internals.sh@116]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 35:47.69)
[35:48] / make[2]: *** [build] Error 1
make[1]: *** [_toolchain] Error 2
make: *** [/Volumes/case-sensitive/esp-open-sdk-3/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

It seems to have worked but failed on a cleanup step.

jacobjennings, 2016-11-05:
seems related to this function in crosstool-NG/scripts/functions

`

# Removes one or more directories, even if it is read-only, or its parent is
# Usage: CT_DoForceRmdir dir [...]
CT_DoForceRmdir() {
    local dir
    local mode
    for dir in "${@}"; do
        [ -d "${dir}" ] || continue
        case "$CT_SYS_OS" in
            Linux|CYGWIN*)
                mode="$(stat -c '%a' "$(dirname "${dir}")")"
                ;;
            Darwin|*BSD)
                mode="$(stat -f '%Lp' "$(dirname "${dir}")")"
                ;;
            *)
                CT_Abort "Unhandled host OS $CT_SYS_OS"
                ;;
        esac
        CT_DoExecLog ALL chmod u+w "$(dirname "${dir}")"
        CT_DoExecLog ALL chmod -R u+w "${dir}"
        CT_DoExecLog ALL rm -rf "${dir}"
        CT_DoExecLog ALL chmod ${mode} "$(dirname "${dir}")"
    done
}

`

jukben, 2018-01-21:
I have the same issue. 

 make toolchain esptool libhal STANDALONE=n

```
[INFO ]  Installing cross-gdb
[INFO ]  Installing cross-gdb: done in 170.13s (at 19:26)
[INFO ]  =================================================================
[INFO ]  Cleaning-up the toolchain's directory
[INFO ]    Stripping all toolchain executables
[19:27] / stat: cannot read file system information for '%Lp': No such file or directory
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Cleaning-up the toolchain's directory'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoForceRmdir[scripts/functions@460]
[ERROR]  >>        called from: do_finish[scripts/build/internals.sh@116]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 19:26.71)
[19:27] / make[2]: *** [build] Error 1
make[1]: *** [_toolchain] Error 2
make: *** [/Volumes/case-sensitive/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
```