pfalcon/esp-open-sdk #223: make STANDALONE=y ERROR

skylin008, 2016-11-10:
When Build the toolchain ,the ERROR information as below:
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_isl_get[scripts/build/companion_libs/121-isl.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@591]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 6:39.59)
[06:48] / make[2]: *** [build] Error 1
make[2]: Leaving directory `/opt/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/opt/esp-open-sdk/crosstool-NG'
make: *** [/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

How I solve this issue?Thank you very much!

carterw, 2016-11-11:
Seeing that here as well. Apparently resulting from many failed attempts to obtain cloog-0.18.4 similar to the first couple of attempts shown below in the build.log;

[ALL  ]    --2016-11-10 17:52:58--  http://www.bastoul.net/cloog/pages/download/cloog-0.18.4.tar.xz
[ALL  ]    Resolving www.bastoul.net (www.bastoul.net)... 62.210.16.62
[ALL  ]    Connecting to www.bastoul.net (www.bastoul.net)|62.210.16.62|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 404 Not Found
[ALL  ]    2016-11-10 17:52:58 ERROR 404: Not Found.
[ALL  ]    
[DEBUG]    Not at this location: "http://www.bastoul.net/cloog/pages/download/cloog-0.18.4.tar.xz"
[DEBUG]    Trying 'ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.4.tar.xz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/home/bill/esp-open-sdk/crosstool-NG/.build/tarballs/cloog-0.18.4.tar.xz.tmp-dl' 'ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.4.tar.xz' 
[ALL  ]    --2016-11-10 17:52:58--  ftp://gcc.gnu.org/pub/gcc/infrastructure/cloog-0.18.4.tar.xz
[ALL  ]               => '/home/bill/esp-open-sdk/crosstool-NG/.build/tarballs/cloog-0.18.4.tar.xz.tmp-dl'
[ALL  ]    Resolving gcc.gnu.org (gcc.gnu.org)... 209.132.180.131
[ALL  ]    Connecting to gcc.gnu.org (gcc.gnu.org)|209.132.180.131|:21... connected.
[ALL  ]    Logging in as anonymous ... Logged in!
[ALL  ]    ==> SYST ... done.    ==> PWD ... done.
[ALL  ]    ==> TYPE I ... done.  ==> CWD (1) /pub/gcc/infrastructure ... done.
[ALL  ]    ==> SIZE cloog-0.18.4.tar.xz ... done.
[ALL  ]    
[ALL  ]    ==> PASV ... done.    ==> RETR cloog-0.18.4.tar.xz ... 
[ALL  ]    No such file 'cloog-0.18.4.tar.xz'

skylin008, 2016-11-11:
Than you carterw! I will be try!

carterw, 2016-11-11:
I'm using Ubuntu Trusty running in a virtual machine. I tried manually installing cloog;
sudo apt-get install cloog-isl

Which succeeded, but I still get the log errors. The script attempts to pull it in anyways. What's odd is that I can get to the sites that are being tried via command line, no obvious reason why the build script can't get to them.

carterw, 2016-11-12:
I came up with a workaround for this problem. 

The build scripts appear to be looking for cloog tarballs that have a .tar.xz extension for some reason, whereas the current tarballs are .tar.gz. I couldn't see where the extension was specified, but I did discover that the tarballs are extracted into esp-open-sdk/crosstool-NG/.build/src. If you perform this extraction manually the script won't attempt to fetch them and the problem is bypassed.

So, cd into the above directory and invoke;
wget --passive-ftp -nc --progress=dot:binary -T 10 http://www.bastoul.net/cloog/pages/download/cloog-0.18.4.tar.gz

This should pull in the cloog tarball, and then you can;
tar xf cloog-0.18.4.tar.gz

That results in a cloog-0.18.4 directory, and the script is satisfied. A number of libraries are pulled in as source like this, and later built and linked directly into the toolchain. I don't know why they didn't just include all that stuff in the already lengthy list of installed prerequisites and simply link to those resulting libraries.

niallp, 2017-02-26:
It looks like the download problem is getting worse, bastoul.net has been offline the last couple of days. I found another copy of a cloog tarball on github but it doesn't seem to be compatible, it builds but fails further down.

Could anyone provide a copy of cloog-0.18.4.tar.gz from their tarballs directory ? I've tried contacting the admin for bastoul.net, hopefully a temporary issue but still dead today.

thanks.

jeremyd2019, 2017-02-26:
> Could anyone provide a copy of cloog-0.18.4.tar.gz from their tarballs directory ?

I uploaded all of the tarballs to the release I made on my fork to hold my mingw32 build.

[https://github.com/jeremyd2019/crosstool-NG/releases/tag/xtensa-1.22.x-g46f160f](https://github.com/jeremyd2019/crosstool-NG/releases/tag/xtensa-1.22.x-g46f160f)

HectorBenito, 2017-09-27:
Wow this worked! It was driving me crazy. But in my case the error was with expat-2.1.0.tar.gz