pfalcon/esp-open-sdk #224: make flash  [Errno 2] No such file or directory: 'blinky-0x40000.bin'

CorrederaJorge, 2016-11-13:
Hello,

I have build the tool using:

`make STANDALONE=y`

 I am able to compile the example project blinky using make:
```
xtensa-lx106-elf-gcc -I. -mlongcalls   -c -o blinky.o blinky.c
xtensa-lx106-elf-gcc -Teagle.app.v6.ld  blinky.o  -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -lc -Wl,--end-group -lgcc -o blinky
esptool.py elf2image blinky
esptool.py v1.2
```
 
However it generates blinky-0x00000.bin  blinky-0x10000.bin . When I try to flash using make file it requires blinky-0x400000.bin and of course it not generated. 

What am I doing wrong?

hieuvt6-fti, 2016-11-18:
Ah, I didn't notice your complaint when I open my similar issue.

You need to replace all "0x40000" in the Makefile with "0x10000".
Apparently the example Makefile expected the compiler to generate a blinky-0x40000.bin, but the compiler generated a blinky-0x10000.bin.

CorrederaJorge, 2016-11-19:
Please close this issue as repeated #226

closed 2016-11-19 by CorrederaJorge