pfalcon/esp-open-sdk #226: Makefile for blinky example need update

hieuvt6-fti, 2016-11-18:
I just freshly installed the esp-open-sdk on a freshly installed Ubuntu virtual machine. Attempting to compile the blinky example, I found that I need to modify the Makefile into following for it to work:

```
CC = xtensa-lx106-elf-gcc
CFLAGS = -I. -mlongcalls
#LDLIBS = -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -Wl,--end-group -lgcc
LDLIBS = -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -lc -Wl,--end-group -lgcc
LDFLAGS = -Teagle.app.v6.ld

blinky-0x00000.bin: blinky
	esptool.py elf2image $^

blinky: blinky.o

blinky.o: blinky.c

flash: blinky-0x10000.bin
	esptool.py write_flash 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

clean:
	rm -f blinky blinky.o blinky-0x00000.bin blinky-0x10000.bin

```

The modification to LDLIBS is required to resolve an issue where make fail with no reference to "memchr()." Otherwise, the toolchain generate a 0x00000.bin and 0x10000.bin instead of 0x00000.bin and 0x40000.bin like the Makefile is designed to work with, so I need to change all 0x40000 to 0x10000 in the makefile.

I am inexperienced with the ESP Open SDK and Linux in general, so if there is a better way to do this, please do let me know

CorrederaJorge, 2016-11-19:
Hi,

Thank you for your answer. Could you explain why the new esptool is generating 0x10000.bin instead of the old 0x40000.bin ?

hieuvt6-fti, 2016-11-21:
Sorry, CorrederaJorge, I do not know. As I mentioned I am new to this entire ESP8266 system myself.

And as a matter of fact, I also want to know why and how esptool is generating such files. I don't feel comfortable not knowing how a toolchain work.

Anyhow it did took me a fair amount of Googling to figure out what is wrong, so I wanted to open an issue here in hope some dev get it fixed for the future users.

CorrederaJorge, 2016-11-26:
Could you say me what is the expected behaviour of the esp after flashing this program?

hieuvt6-fti, 2016-11-26:
It toggles GPIO1s every 500ms.
You should look into the source code. There is more info there.

larsch, 2016-12-01:
I'm trying out the current esp-open-sdk. The blinky example Makefile has the same issues as mentioned here (The esptool.py generates the 0x10000 image instead of 0x40000). When flashed, my LED doesn't blink though - and the board tends to reset a lot. When I compile from commit 9dac805 from 4 months ago, the blinky example works fine.

alexwilburphilips, 2016-12-04:
@larsch Been fighting with this today as well. I have a LoLin "NodeMCU v3" board, which appears to have the built-in (blue) LED on GPIO2. This LED is the one that sits below the ESP8266 module's wifi antenna.

Here are the changes I had to make:

A) In the blinky makefile:
1) I edited the LDLIBS line to add the missing "-lc" argument (as per this issue)
2) Edited the flash section to read:

esptool.py -p /dev/tty.wchusbserial1410 write_flash 0x00000 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

B) In the blinky.c file:
3) Changed the pin line to:

static const int pin = 2;

This now appears to work as I'd expect. I also have an Adafruit huzzah and this works exactly the same there too, I just need to change the port parameter as passed to esptool.py.

bobo1on1, 2016-12-07:
The reason that esptool now generates 0x10000.bin instead of 0x40000.bin is because this change in eagle.app.v6.ld:

old:
  irom0_0_seg :                         org = 0x40240000, len = 0x3C000

new:
  irom0_0_seg :                         org = 0x40210000, len = 0x5C000

lmayainsite, 2016-12-10:
Is it safe to use 0x10000 then?, we are fighting the same situation with blinky and hello world.

dodoma, 2016-12-13:
0x10000 works fine on my ESP-12F

pudumula, 2017-02-26:
Hey guys can I make the required changes to the make file and issue a pull request ?

jcmvbkbc, 2017-02-27:
> Hey guys can I make the required changes to the make file and issue a pull request ?

Why do you even ask?

Navis-Raven, 2017-07-20:
Why this issue is still active and hasn't been not corrected yet ? 
I encountered this issue just yesteray while trying to solve another issue.

Navis-Raven, 2017-07-20:
@pudumula  do it

pfalcon, 2017-07-20:
Was fixed by https://github.com/pfalcon/esp-open-sdk/commit/c416ce38f843025401ab3f5d9ca21dad9ccc3f78

closed 2017-07-20 by pfalcon

Navis-Raven, 2017-07-21:
NB: this code has not to be confused with this one https://github.com/esp8266/source-code-examples/tree/master/blinky

mjcross, 2018-01-19:
Very happy to confirm that I've just built _examples/blinky_ using the esp-open-sdk toolchain (STANDALONE=n on Mac OSX High Sierra 10.13.2), uploaded it to an ESP-12F, and I have a nice flashing LED.

Thanks to all contributors above for sharing their knowledge.

For reference the key steps I followed were:

1. fully erase the device and restore the system parameter area:
```
esptool.py --port <serial port> erase_flash
esptool.py --port <serial port> write_flash 0x3FC000 esp_init_data_default.bin
```
(NB: as noted by @sambazley the address of 0x3FC000 depends on the size of your flash - see [https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/flash.md#sdk-init-data](sdk-init-data)

2. upload the application:
`esptool.py --port <serial port> write_flash 0x00000 blinky-0x00000.bin 0x10000 blinky-0x10000.bin`

I didn't need to specify the flash_mode or flash_freq for esptool.py, I just let it decide for me (esptool.py v2.2, installed with pip).

anthcp, 2018-07-10:
Just tried the above on my NodeMCU ESP8266 and it does not work... But Adurino ESP8266 and MicroPython work fine...
How can I help here?

Navis-Raven, 2018-08-30:
@anthcp So it seems that it is not the same issue. You should open another issue. The above is for esp-open-sdk