pfalcon/esp-open-sdk #228: region `iram1_0_seg' overflowed by 1000 bytes

GreatProjects, 2016-12-01:
If i try to build my project i get this output:
`xtensa-lx106-elf-gcc -I. -mlongcalls   -c -o segmentled.o segmentled.c
xtensa-lx106-elf-gcc -Teagle.app.v6.ld  segmentled.o  -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -lc -Wl,--end-group -lgcc -o segmentled
/home/test/Desktop/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: segmentled section `.text' will not fit in region `iram1_0_seg'
/home/test/Desktop/opt/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: region `iram1_0_seg' overflowed by 1000 bytes
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'segmentled' failed
make: *** [segmentled] Error 1
`
Even though my code is not very long (just about 200 lines). I added "ICACHE_FLASH_ATTR" in front of every function. Do you have any idea, how i can fix this error? Any help would be very much appreciated.

cranphin, 2016-12-02:
I expect the actual code would help here.

GreatProjects, 2016-12-02:
Ok, here is my code:

http://pastebin.com/NaCJx1ZF

pepe79, 2016-12-17:
I fixed the same error by adding  -DICACHE_FLASH to the CFLAGS variable in the Makefile:
`CFLAGS = -I. -DICACHE_FLASH -mlongcalls
`
as suggested in https://github.com/pfalcon/esp-open-sdk/issues/82

GreatProjects, 2016-12-17:
Thanks for your reply, sadly, this didn't do anything. I still get the same error :(

pepe79, 2016-12-17:
You forgot the minus sign before DICACHE_FLASH. I tested your code with my Makefile and it works (without -DICACHE_FLASH it doesn't).

GreatProjects, 2016-12-17:
Thanks a lot! Now everything works :)

closed 2017-05-21 by pfalcon