pfalcon/esp-open-sdk #231: Error installing crosstool-NG MakeFile pointing at `pwd`

rsegecin, 2016-12-18:
When building the crosstool-NG through the command "make STANDALONE=y" it throws the error below. Where the prefix should be pointed at? if the case it should be on crosstool-NG directory so ./configure would be enough. So I've altered the makefile to just `./configure` but then I get an error saying that I should grant root permition to add an file called '/usr/local/bin/ct-ng' when performing `make install MAKELEVEL=0`. Did you have to grant root privileges to install crosstool-NG? 

`...
./configure --prefix=`pwd`
configure: WARNING: you should use --build, --host, --target
configure: WARNING: invalid host type: Projects/esp-open-sdk/crosstool-NG
checking build system type... Invalid configuration `Projects/esp-open-sdk/crosstool-NG': machine `Projects/esp-open-sdk/crosstool' not recognized
configure: error: /bin/bash scripts/config.sub Projects/esp-open-sdk/crosstool-NG failed
...`

jcmvbkbc, 2016-12-18:
> it throws the error below

Please provide the full error message. Preferably the whole build log.

> Did you have to grant root privileges to install crosstool-NG

No. All changes made during toolchain build are local to the esp-open-sdk directory and none of them requires root privileges.

rsegecin, 2016-12-18:
Thank you for the quick reply jcmvbkbc, I'm honored to be speaking with you. I've just finished installing the esp-open-sdk the error I was getting was because the directory path to the esp-open-sdk had a space in one of the name directories it was under losing the reference. Sorry it was a silly thing but I hope anyone ever get stuck on the same problem will be able to find the solution here.

closed 2016-12-18 by rsegecin