pfalcon/esp-open-sdk #235: Diferences between ESP8266_RTOS_SDK and esp-open-sdk

rsegecin, 2017-01-08:
Hi. I've compile and flashed both examples from ESP8266_RTOS_SDK (using gen_misc.sh) and esp-open-sdk both I noticed uses different libraries. esp-open-sdk seems to point to nonOS libraries but works with RTOS as well?! There's the blink project template that I've tried to include the uart library `#include "uart.h"` but it doesn't find it so I've tried to copy and paste it on the project (I don't think that's the right way) but it can't find the function `UART_SetBaudrate`, how should I properly add the uart lib? Should I compile with `gen_misc.sh` or using the makefile in the blink project?

pfalcon, 2017-01-08:
> esp-open-sdk seems to point to nonOS libraries but works with RTOS as well?!

In general, no.

Can't help much with the other question, sorry (other people might). I just can make a generic suggestion to pause for a while and think what exactly you want to do, and why you're trying to do it that way. esp-open-sdk's README gives a reference to a pretty big project which uses esp-open-sdk, which has code to deal with UART too (that's if you'd come to conclusion that you need esp-open-sdk, not something else, make you sure you know about https://github.com/SuperHouse/esp-open-rtos).

rsegecin, 2017-01-08:
I just found it confusing because the blinky example implements os_timer_t that I thought only a RTOS would have, so I didn't know what would be better to compile my application with. For know I really don't have a preference whether to use RTOS or not, I just want the latest sdk that have more features and less bugs. I saw the esp-open-rtos but it's based on Espressif RTOS SDK 0.9.9 and I'd like to work with more recent sdk that brought optmizations to mDNS that my application is gonna use. Their version has more of memory management improvements I'm skeptical about the rest because I didn't find any clue.

pearson, 2017-01-26:
The os_timer_t isn't working without the RTOS. The poor LED on my Huzzah Feather board is flashing extremely quickly - almost to the point of not being noticeable.  It's definitely not a 500ms delay.

ZyperPL, 2017-07-05:
@pfalcon Does you or someone else know why esp-open-sdk doesn't implement UART, but Espressif Non OS SDK does?
At least that's what I can read in their ESP8266 Non OS SDK API Reference.