pfalcon/esp-open-sdk #236: build error: conflicting types for 'kconf_id_lookup'

aifer2007, 2017-01-09:
In file included from zconf.tab.c:237:
./zconf.hash.c:167:1: error: conflicting types for 'kconf_id_lookup'
kconf_id_lookup (register const char *str, register size_t len)
^
./zconf.hash.c:34:31: note: previous declaration is here
static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                              ^
1 error generated.
make: *** [zconf.tab.o] Error 1
make: *** [build-lib-kconfig] Error 2
make: *** [build] Error 2
make[1]: *** [_ct-ng] Error 2
make: *** [crosstool-NG/ct-ng] Error 2

Dennis-Janssen, 2017-01-10:
I get the same error on macOS 10.12.2.

aifer2007, 2017-01-10:
my Xcode : Version 8.2.1 (8C1002)

Dennis-Janssen, 2017-01-10:
It might have something to do with a recent update to gperf a few days ago after having 8 years of no updates.
Been trying to find a way to install the previous version using brew but no luck so far.

aifer2007, 2017-01-10:
@Dennis-Janssen

or，you can install macports, and use

$ sudo port install git gsed gawk binutils gperf grep gettext py-serial wget libtool  autoconf automake

instead of :

$ brew install binutils coreutils automake wget gawk libtool help2man gperf gnu-sed --with-default-names grep

It works well!

Reference:

https://github.com/pfalcon/esp-open-sdk/issues/11

rversteegen, 2017-02-22:
Fixed upstream:
https://github.com/crosstool-ng/crosstool-ng/issues/528