pfalcon/esp-open-sdk #237: Retreiving neede toolchain components' tarballs fails

cpranzl, 2017-01-15:
Hi,
trying to build micropython myself and running repeatly into this error

```
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/home/cpranzl/Projekte/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Verzeichnis „/home/cpranzl/Projekte/esp-open-sdk/crosstool-NG“ wird betreten
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20170115.163602
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[04:02] / 

[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_isl_get[scripts/build/companion_libs/121-isl.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@591]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-55-gecfc19a/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 5:28.80)
[05:29] / ct-ng:152: die Regel für Ziel „build“ scheiterte
make[2]: *** [build] Fehler 1
make[2]: Verzeichnis „/home/cpranzl/Projekte/esp-open-sdk/crosstool-NG“ wird verlassen
../Makefile:130: die Regel für Ziel „_toolchain“ scheiterte
make[1]: *** [_toolchain] Fehler 2
make[1]: Verzeichnis „/home/cpranzl/Projekte/esp-open-sdk/crosstool-NG“ wird verlassen
Makefile:126: die Regel für Ziel „/home/cpranzl/Projekte/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc“ scheiterte
make: *** [/home/cpranzl/Projekte/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Fehler 2
cpranzl@thor:~/Projekte/esp-open-sdk$7
```
KR
Christoph

cpranzl, 2017-01-15:
Output is in German but it tells me that there is a problem with retrieving tarball for crosstool-NG. I tried to clone the crosstool-NG project which works fine.

sheinz, 2017-01-15:
I have a similar issue.
In my case the problem is that a server "http://isl.gforge.inria.fr/isl-0.14" is not responding.
You can check the more detailed info in file `crosstool-NG/build.log`

```bash
cat crosstool-NG/build.log

[ALL  ]    Retrying.
[ALL  ]
[ALL  ]    --2017-01-15 17:30:24--  (try: 3)  http://isl.gforge.inria.fr/isl-0.14
[ALL  ]    Connecting to isl.gforge.inria.fr (isl.gforge.inria.fr)|128.93.193.15|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
[ALL  ]    Giving up.
[ALL  ]
[DEBUG]    Not at this location: "http://isl.gforge.inria.fr/isl-0.14"
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
```

nubcore, 2017-01-16:
I'm getting the same thing, just re-created a fresh build environment, with same sources as yesterday, and now it's hanging on downloading the tarballs

nubcore, 2017-01-16:
it looks like the server that wasn't responding, is now responding.

the tarballs get downloaded to /esp-open-sdk/crosstool-NG/.build/tarballs/ I was just now able to delete those, run "make clean" and then "make", all the tarballs were successfully downloaded this time, and compile. So alternatively you could download and place them manually from http://isl.gforge.inria.fr/

```
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20170115.224951
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 87.72s (at 01:29)

```

also noticed there are newer versions available, and potentially alternatively retrievable from git repo http://repo.or.cz/w/isl.git

cpranzl, 2017-01-16:
I can confirm that the problem is now solved.

closed 2017-01-16 by cpranzl

yadavravi079, 2017-08-16:
To get rid of this Error, try to fix FTP proxy

cemox, 2018-03-04:
goto crosstool-NB/.build/tarballs directory and download expat-2.1.0 by running:

`wget https://github.com/libexpat/libexpat/releases/download/
R_2_1_0/expat-2.1.0.tar.gz
`

mh4nx7net, 2020-10-08:
https://master.dl.sourceforge.net/project/expat/expat/2.2.10/expat-2.2.10.tar.xz

SinanAkkoyun, 2021-05-04:
For anyone living in 2021 and needing it for esp-open-sdk:
https://src.fedoraproject.org/repo/pkgs/expat/expat-2.1.0.tar.gz/dd7dab7a5fea97d2a6a43f511449b7cd/

Idk if this is a virus but it worked for me

raspiduino, 2021-07-19:
> goto crosstool-NG/.build/tarballs directory and download expat-2.1.0 by running:
> 
> `wget https://github.com/libexpat/libexpat/releases/download/ R_2_1_0/expat-2.1.0.tar.gz `

It should be
`wget https://github.com/libexpat/libexpat/releases/download/R_2_1_0/expat-2.1.0.tar.gz `

Bohan-Song, 2021-08-25:
still error

0z4, 2021-09-01:
> still error

yep, im going to do a PR with an answere above that worked

dsotirakis, 2021-10-04:
Any ideas on how can we poke people about http://isl.gforge.inria.fr/ being down?

adolf-wv, 2021-10-09:
> Any ideas on how can we poke people about http://isl.gforge.inria.fr/ being down?

You can use wayback machine to fetch the isl library until the site comes back up (https://web.archive.org/web/20210701145459/http://isl.gforge.inria.fr/isl-0.14.tar.gz) and add it to your `crosstool-NG/.build/tarballs` directory

Abeldbest, 2021-10-17:
> You can use wayback machine to fetch the isl library until the site comes back up (https://web.archive.org/web/20210701145459/http://isl.gforge.inria.fr/isl-0.14.tar.gz) and add it to your [...]

I have added (https://web.archive.org/web/20210701145459/http://isl.gforge.inria.fr/isl-0.14.tar.gz) to .build but still fails
[image]

marc-hb, 2021-11-12:
https://libisl.sourceforge.io/ seems to be indeed the new, official ISL site.

In case it ever goes down too or moves again, gcc is mirroring the old ISL versions it depends on:
https://gcc.gnu.org/install/prerequisites.html
https://gcc.gnu.org/pub/gcc/infrastructure/

A local mirror is how every project should isolate itself from external downtimes.

ChrisMacGregor, 2022-01-12:
I subsequently ran into a couple of other issues that needed patches.  The end result (which gets me a clean build on Ubuntu 21.10 of an apparently working toolchain) can be found here:

https://github.com/pfalcon/esp-open-sdk/pull/391
https://github.com/ChrisMacGregor/esp-open-sdk/tree/builds-on-Ubuntu-21.10
Tarball of built toolchain: https://github.com/ChrisMacGregor/esp-open-sdk/releases/tag/builds-on-Ubuntu-21.10

Jibun-no-Kage, 2022-08-14:
Nothing above seems to work, completely broken at this point.  Help!

astuaTEC, 2023-03-30:
> Had errors with both `isl` and `expat`. From `esp-open-sdk` dir, did this (after a failed `make`):
> 
> ```shell
> cd crosstool-NG/.build/tarballs
> wget https://libisl.sourceforge.io/isl-0.14.tar.gz
> wget https://github.com/libexpat/libexpat/releases/download/R_2_1_0/expat-2.1.0.tar.gz
> cd -
> make
> ```
> 
> Seems to be compiling successfully now.

Now (03/2023) you need to use these links:

> ```shell
> cd crosstool-ng/.build/tarballs
> wget https://libisl.sourceforge.io/isl-0.20.tar.gz
> wget https://github.com/libexpat/libexpat/releases/download/R_2_2_6/expat-2.2.6.tar.bz2
> cd -
> make
> ./ct-ng build
> ```