pfalcon/esp-open-sdk #239: Cannot download cloog / new cloog source

pearson, 2017-01-25:
First, big kudos for your build script(s). It's one of the most robust I've ever seen. 

That said, it broke for me. :-) Unfortunately the sources of cloog-0.18.4 in the scripts either don't have it (ftp://gcc.gnu.org/pub/gcc/infrastructure/) or are currently offline (http://www.bastoul.net/). As I'm new here, I don't know how reliable bastoul.net is.

Regardless, here's another source of cloog-0.18.4. The only difficulty is the different name than what is expected. (It expects cloog_0.18.4.tar.gz and this is cloog_0.18.4.**orig**.tar.gz) A bit of hacking at the scripts did get it to download, build, and use this version, though. Presumably someone who has a better knowledge of the scripts could integrate this source.

http://archive.ubuntu.com/ubuntu/pool/universe/c/cloog/cloog_0.18.4.orig.tar.gz

Presumably archive.ubuntu.com will be reliable. :-)

pfalcon, 2017-02-07:
Please see https://github.com/pfalcon/esp-open-sdk/issues/100.

closed 2017-02-07 by pfalcon

pearson, 2017-02-14:
It's probably worth noting that both sources in the makefile for cloog are still unavailable. Perhaps adding the Ubuntu mirror would be a good idea?