pfalcon/esp-open-sdk #243: make error: zconf.hash.c:167:1: error: conflicting types for ‘kconf_id_lookup’

markusvanlaak, 2017-02-19:
Hi,

Arch Linux. Cloned the repository, run make and get:

___
zconf.hash.c:167:1: error: conflicting types for ‘kconf_id_lookup’
 kconf_id_lookup (register const char *str, register size_t len)
 ^~~~~~~~~~~~~~~
zconf.hash.c:34:31: note: previous declaration of ‘kconf_id_lookup’ was here
 static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                               ^~~~~~~~~~~~~~~
---------------
Probably the same issue with the version conflict of gperf described in other open issues.
I run version 3.1:
-----
#gperf -v
GNU gperf 3.1
Copyright (C) 1989-2017 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Written by Douglas C. Schmidt and Bruno Haible

Someone run into the same issue and fixed it?

Thanks for your support.
Markus

pedrominatel, 2017-02-25:
Same issue on Manjaro.

rillhu, 2017-03-02:
Same issue on MACOS 10.12

gperf -v
GNU gperf 3.1
Copyright (C) 1989-2017 Free Software Foundation, Inc.

make

./zconf.hash.c:167:1: error: conflicting types for 'kconf_id_lookup'
kconf_id_lookup (register const char *str, register size_t len)
^
./zconf.hash.c:34:31: note: previous declaration is here
static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);

Aterocana, 2017-03-03:
Same here on Archlinux.

riban-bw, 2017-03-04:
Same on msys2 on Windows 7 64-bit.

rillhu, 2017-03-06:
I copy the solution from issue #236 👍 
Just uninstalling the brew version of **gperf (3.1)** and let the OSX version (**3.0.3**) take over seems to do the trick.

With above solution I can build the sdk on my Mac

Quanalogy, 2017-03-06:
similar issue here:
In file included from zconf.tab.c:213:0:
zconf.hash.c:163:1: error: conflicting types for ‘kconf_id_lookup’
 kconf_id_lookup (register const char *str, register size_t len)

Arch linux

TreehouseAndSky, 2017-03-14:
As @rillhu commented, deleted the brew version and that did it.

zoobab, 2017-03-20:
Same issue on my sabayon gentoo.

Aterocana, 2017-03-20:
As said before: just downgrade your gperf package to previous version.
gperf 3.0.4 works fine for me.

Aterocana, 2017-03-20:
> Same issue on my sabayon gentoo.

As said before: just downgrade your gperf package to previous version.
gperf 3.0.4 works fine for me.

zoobab, 2017-03-20:
yeah worked.

riban-bw, 2017-03-20:
Some platforms do not provide a downgrade path, e.g. I do not know how to install earlier version of gpref on MSYS2. There is a fix for this upstream and it would be advantageous for esp-open-sdk to update to a leter version of crosstool-NG. Meanwhile, if anyone knows how to install / downgrade to earlier version of gpref on MSYS2, please tell me.

gromain, 2017-04-03:
Applying this upstream patch directly to crosstools-NG worked for me: https://github.com/crosstool-ng/crosstool-ng/pull/529/commits/3bb2bcd2a328fb56332398783e91899ce3bc3556

I'm using Manjaro.

riban-bw, 2017-04-03:
@gromain that patch is against a different version of crosstool-ng to the one currently used by esp-open-sdk. The patch does not cleanly apply to this version. Can @pfalcon change the version of crosstool-ng to allow this fix to be applied? (I know that is easier said than done!)

pfalcon, 2017-04-03:
> (I know that is easier said than done!)

Exactly. Someone would need to prove that compiler built by the new version of crosstool-ng works not worse than built by the one currently in esp-open-sdk, and that's hard.

But guys, by all means feel free to either use crosstool-ng directly, or feel free to update esp-open-sdk's submodule of it, build it, and in few weeks share your experiences.

gromain, 2017-04-03:
@riban-bw Sorry, I wasn't clear.
I applied by hand the patch to the relevant files.

The main problem is in `configure.ac` where the change on lines 137, 138, 139 doesn't apply and where the addition is not at line 307 but at line 293.

I'm sorry my message was confusing.

Here is a complete diff from my hand-modified files:
`diff --git a/Makefile.in b/Makefile.in`
```
index 3df6597a..19ca4e6e 100644
--- a/Makefile.in
+++ b/Makefile.in
@@ -69,6 +69,7 @@ export objdump         := @OBJDUMP@
 export readelf         := @READELF@
 export patch           := @PATCH@
 export gperf           := @GPERF@
+export gperf_len_type  := @GPERF_LEN_TYPE@
 export CC              := @CC@
 export CPP             := @CPP@
 export CPPFLAGS        := @CPPFLAGS@
```

`diff --git a/configure.ac b/configure.ac`
```
index 4f0b4283..5d512fe8 100644
--- a/configure.ac
+++ b/configure.ac
@@ -290,6 +290,22 @@ AC_CACHE_CHECK([for GNU automake >= 1.10], [ac_cv_path_automake],
          AC_MSG_ERROR([could not find GNU automake >= 1.10])])])
 AC_SUBST([automake], [$ac_cv_path_automake])

+#----------------------------------------
+# Gperf 3.1 started generating functions with size_t rather than unsigned int
+AC_MSG_CHECKING([for the type used in gperf declarations])
+echo "#include <string.h>" > conftest.gperf.c
+echo "const char * in_word_set(const char *, GPERF_LEN_TYPE);" >> conftest.gperf.c
+echo foo,bar | ${GPERF} -L ANSI-C >> conftest.gperf.c
+AS_IF([${CC} -c -o /dev/null conftest.gperf.c -DGPERF_LEN_TYPE='size_t' >/dev/null 2>&1],
+            [AC_MSG_RESULT([size_t])
+             GPERF_LEN_TYPE='size_t'],
+        [${CC} -c -o /dev/null conftest.gperf.c -DGPERF_LEN_TYPE='unsigned int' >/dev/null 2>&1],
+            [AC_MSG_RESULT([unsigned int])
+             GPERF_LEN_TYPE='unsigned int'],
+        [AC_MSG_ERROR([unable to determine gperf len type])])
+rm -f conftest.gperf.c
+AC_SUBST([GPERF_LEN_TYPE])
+
 #--------------------------------------------------------------------
 # Boring again... But still a bit of work to do...
 #--------------------------------------------------------------------
```

`diff --git a/kconfig/Makefile b/kconfig/Makefile`
```
index 9f30566a..ee838aa3 100644
--- a/kconfig/Makefile
+++ b/kconfig/Makefile
@@ -6,7 +6,8 @@ all: conf mconf nconf
        @true   # Just be silent, you fscking son of a fscking beach...

 # Build flags
-CFLAGS = -DCONFIG_=\"CT_\" -DPACKAGE="\"crosstool-NG $(VERSION)\""
+CFLAGS = -DCONFIG_=\"CT_\" -DPACKAGE="\"crosstool-NG $(VERSION)\"" \
+        -DGPERF_LEN_TYPE="$(gperf_len_type)"
 LDFLAGS = $(INTL_LIBS)

 # Compiler flags to use gettext
```

`diff --git a/kconfig/zconf.gperf b/kconfig/zconf.gperf`
```
index ac498f01..434483b3 100644
--- a/kconfig/zconf.gperf
+++ b/kconfig/zconf.gperf
@@ -9,7 +9,7 @@

 struct kconf_id;

-static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
+static const struct kconf_id *kconf_id_lookup(register const char *str, register GPERF_LEN_TYPE len);
```

kenshinyao, 2018-01-25:
find the hconf.c file in nuttx

search:
kconf_id_lookup (register const char *str, register size_t len)
**change to**
kconf_id_lookup (register const char *str, register unsigned int len) 

then it works fine for me

bohdan-chupys, 2019-06-05:
> Same here on Archlinux.
> Solved downgrading gperf package using Archlinux Archive to version 3.0.4-7 which you can find here: https://archive.archlinux.org/packages/g/gperf/

Same issue. Also on Arch. 
Little update: 
On the mentioned location I didn`t find version 3.0.4-7.
So I have to build it from source code. The source code I download here:
http://www.linuxfromscratch.org/blfs/view/7.5/general/gperf.html
The building process is simple. 
Downgrade helps me.