pfalcon/esp-open-sdk #248: Error Building In Ubuntu 16.04 LTS crosstool-NG

UlmusLeo, 2017-03-10:
>make[2]: Entering directory '/home/name/Documents/esp-open-sdk/crosstool-NG\'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20170310.121909
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_expat_get[scripts/build/companion_libs/210-expat.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@591]
[ERROR]  >>

This error seems to be because it is trying to reference an old version of expat
The build is successful when the file crosstool-NG/scripts/build/companion_libs/210-expat.sh is renamed to 220-expat.sh which then downloads and installs expat version 2.2.0 with no reported errors.

I am new to esp and am not sure if upgrading the version here breaks anything.

geokon-gh, 2017-03-11:
(@UlmusLeo .. just a sanity check for you)
I'm also having the same issue, though I couldn't get your fix working unfortunately
I also tried on my ARM machine (Chromebook + crouton) and it too fails in crosstool-NG, but at a different companion-lib

Looks like the esp-open-sdk needs an update, but I'm also new to ESP - so it's hard for me to debug the problem as well

supramaterial, 2018-03-01:
A way to fix this issue is this pull request: https://github.com/jcmvbkbc/crosstool-NG/pull/53
This also fixes https://github.com/pfalcon/esp-open-sdk/issues/307

It boils down to update the URL template in crosstool-NG/scripts/build/companion_libs/210-expat.sh from
`http://downloads.sourceforge.net/project/expat/expat/${CT_EXPAT_VERSION}`
to 
`https://github.com/libexpat/libexpat/releases/download/R_${CT_EXPAT_VERSION//[.]/_}`