pfalcon/esp-open-sdk #249: cannot find -ludhcp

atlask1, 2017-03-14:
Hi, I have build esp-open-sdk and if I use 'gen_misc.sh' all is ok.
Now I'm tring to write a makefile but when I try to compile return me this error:
`esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot find -ludhcp
collect2: error: ld returned 1 exit status
Makefile:132: set di istruzioni per l'obiettivo "build/app.out" non riuscito
make: *** [build/app.out] Errore 1
`
Surely it's my fault , but could someone make me understand?
thanks

atlask1, 2017-03-14:
I forgot:
I'm using the last RTOS SDK

pfalcon, 2017-05-21:
There's no gen_misc.sh in esp-open-sdk.

pfalcon, 2017-05-21:
See https://github.com/pfalcon/esp-open-sdk/issues/101

closed 2017-05-21 by pfalcon