pfalcon/esp-open-sdk #251: make fails

mr-bandit, 2017-03-28:
Make fails

Using a Cygwin from 2012

Sequence:
got the git .zip files
did the "git clone --recursive ......" sequence
make (from the top directory)
Output below

**Makefile:181: warning: overriding commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:176: warning: ignoring old commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
fatal: Not a git repository (or any of the parent directories): .git
make: *** [crosstool-NG/bootstrap] Error 128**

If I commented out the line that tried to do the "git submodule update", got another failure: it did not cd to a subdirectory, and got the message:

**make -C crosstool-NG -f ../Makefile _ct-ng
make[1] Entering directory (my directory tree)/freertos/esd-open-sdk/crosstool-NG
./bootstrap
/bin/bash: ./bootstrap: No such file or directory
make[1]: Leaving directory (my directory tree)/freertos/esd-open-sdk/crosstool-NG**

Also, following the instructions:

**git pull**

responds with "**fatal: Not a git repository (or any of the parent directories): .git**"

_Bottom line: the build sequence should be cookbook_

sufex00, 2017-04-08:
Using instructions for Ubuntu 14.04

sudo apt-get install make unrar-free autoconf automake libtool gcc g++ gperf
flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial
sed git unzip bash help2man wget bzip2

and then running make however fixed the issue.

wgreybe, 2017-06-20:
Im trying to setup the OTA client and server on a Rasberry PI3.  When I run the make command (using different options), I get the below error.  I'd appreciate any input.
make ../../esptool/yaota8266-master/
Makefile:187: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:178: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
make: Nothing to be done for '../../esptool/yaota8266-master/'.