pfalcon/esp-open-sdk #252: make fails "/scripts/scripts.mk: No such file or directory"

marduino, 2017-04-02:
make STANDALONE=n
Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16              _08_10/.dir'
Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_              16_08_10/.dir'
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
make -C crosstool-NG -f ../Makefile _toolchain
make[1]: Entering directory '/cygdrive/c/embedded/esp8266/esp-open-sdk/crosstool              -NG'
../Makefile:181: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0              _16_08_10/.dir'
../Makefile:176: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0              .0_16_08_10/.dir'
./ct-ng xtensa-lx106-elf
make[2]: Entering directory '/cygdrive/c/embedded/esp8266/esp-open-sdk/crosstool              -NG'
ct-ng:118: /cygdrive/c/embedded/esp8266/esp-open-sdk/crosstool-NG/lib/crosstool-              ng-1.22.0-55-gecfc19a/scripts/scripts.mk: No such file or directory
make[2]: *** No rule to make target '/cygdrive/c/embedded/esp8266/esp-open-sdk/c              rosstool-NG/lib/crosstool-ng-1.22.0-55-gecfc19a/scripts/scripts.mk'.  Stop.
make[2]: Leaving directory '/cygdrive/c/embedded/esp8266/esp-open-sdk/crosstool-              NG'
make[1]: *** [../Makefile:130: _toolchain] Error 2
make[1]: Leaving directory '/cygdrive/c/embedded/esp8266/esp-open-sdk/crosstool-              NG'
make: *** [Makefile:127: /cygdrive/c/embedded/esp8266/esp-open-sdk/xtensa-lx106-              elf/bin/xtensa-lx106-elf-gcc] Error 2

marduino, 2017-04-09:
I am using cygwin on windows, according with some guidance I have changed"HKLM\SYSTEM\CurrentControlSet\Control\Session Manager\kernel\ dword:obcaseinsensitive", value to '0'

But does not work, is anyone using cygwin? Thanks very much!

davydnorris, 2017-04-24:
The order of things is important here. You have to first set this registry value, then reboot, and then install Cygwin from scratch (because it needs case sensitivity too).

Finally, if none of that works, then you can use the posix=1 option in your Cygwin fstab to force case sensitivity for anything referenced via /cygdrive. Then make your build path relative to that.

marduino, 2017-04-26:
Thanks @davydnorris for the great answer, I will try and get back to you later.

StevoKeano, 2017-11-07:
VI'd my /etc/fstab  and restarted Cygwin; but, no joy.     

none /cygdrive cygdrive binary,posix=1,user 0 0
[image]

davydnorris, 2017-11-07:
@StevoKeano did you set the registry value, then install Cygwin, then set the fstab up?

All three steps are required and I found I had to reinstall Cygwin to get it to recognise case sensitivity as well

StevoKeano, 2017-11-07:
thanks, 
At your suggestion doing all 3:
  1 set dword:obcaseinsensitive to 0
  2 ran setupj-x86.exe and selected Reinstall under "category" Devel
  3 found original /etc/fstab   setting posix=1 

It's Extracting and patching toolchain components

Thank you for the nudge!

davydnorris, 2017-11-07:
Fingers crossed for you!!

StevoKeano, 2017-11-07:
Few errors.... But got past the case sensitive one.

[INFO ]  Installing pass-2 core C gcc compiler
[ERROR]    xgcc: error: spawn: No such file or directory
[INFO ]  Installing pass-2 core C gcc compiler: done in 469.87s (at 36:46)
[INFO ]  =================================================================
[INFO ]  Installing C library
[INFO ]  Installing C library: done in 205.65s (at 40:12)
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[ERROR]    xgcc: error: spawn: No such file or directory
[INFO ]  Installing final gcc compiler: done in 691.92s (at 51:44)
[INFO ]  =================================================================
[INFO ]  Installing cross-gdb
[ERROR]    configure: error: python is missing or unusable
[ERROR]    make[4]: *** [Makefile:8729: configure-gdb] Error 1
[ERROR]    make[3]: *** [Makefile:849: all] Error 2
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 57:54.02)
[57:57] / make[2]: *** [ct-ng:152: build] Error 2
make[2]: Leaving directory '/home/steve/esp-open-sdk/crosstool-NG'
make[1]: *** [../Makefile:137: _toolchain] Error 2
make[1]: Leaving directory '/home/steve/esp-open-sdk/crosstool-NG'
make: *** [Makefile:130: /home/steve/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

steve@i7w10 ~/esp-open-sdk
$ which python
/usr/bin/python

steve@i7w10 ~/esp-open-sdk
$  /usr/bin/python
Python 2.7.14 (default, Oct 31 2017, 21:15:21)
[GCC 6.4.0] on cygwin
Type "help", "copyright", "credits" or "license" for more information.