pfalcon/esp-open-sdk #254: cross-gdb build fails

horvathp, 2017-04-07:
On OS X, with the latest master,
[ERROR]    /Volumes/ESPToolchain/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:231:47: error: incomplete definition of type 'struct scm_t_port'

Is this a known issue?

b3niup, 2017-05-02:
I have very similar issue on linux:
```[ERROR]    /home/XXX/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:231:47: error: dereferencing pointer to incomplete type 'scm_t_port {aka struct scm_t_port}'
[ERROR]    /home/XXX/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:681:19: error: 'SCM_PORT_NEITHER' undeclared (first use in this function)
[ERROR]    /home/XXX/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:711:19: error: 'SCM_PORT_NEITHER' undeclared (first use in this function)
[ERROR]    /home/XXX/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:790:24: error: 'SCM_PORT_WRITE' undeclared (first use in this function)
[ERROR]    /home/XXX/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:816:29: error: 'SCM_PORT_READ' undeclared (first use in this function)
[ERROR]    make[5]: *** [Makefile:2469: scm-ports.o] Error 1
[ERROR]    make[4]: *** [Makefile:8755: all-gdb] Error 2
[ERROR]    make[3]: *** [Makefile:845: all] Error 2
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 13:22.15)

```

little-dude, 2017-05-22:
Same. It seems to be due to this: https://sourceware.org/bugzilla/show_bug.cgi?id=21104

flannelhead, 2017-05-29:
A reasonable workaround to get the rest of the build to succeed is disabling the `gdb` build. It can be done by adding the line
```
CT_DEBUG_gdb=n
```
to the file `crosstool-config-overrides`. A proper workaround would be to somehow pass `--with-guile=guile-2.0` to the `gdb` configure script. I'll report if I figure out how to do that.

flannelhead, 2017-05-29:
Yep, it seems
```
CT_GDB_CROSS_EXTRA_CONFIG_ARRAY="--with-guile=guile-2.0"
```
did the trick and I was able to build also `gdb`. Ideally this situation should be detected on the `crosstool-ng` side, but this serves as a temporary workaround.

Notice that this might break the build for users with older guile. Guile 2.0 has been released in 2011 though, so this shouldn't be a major issue.

roger-zacharczyk, 2017-11-15:
Where did you put this line?

flannelhead, 2017-11-15:
In `crosstool-config-overrides` (which I mentioned in my comment right above).

ricardoquesada, 2018-11-08:
sorry, but where should I put that file, I couldn't find it. I'm following these steps:
https://docs.espressif.com/projects/esp-idf/en/stable/get-started/macos-setup-scratch.html

but I couldn't find any `crosstool-config-overrides` in the `~/esp` folder.
Thanks!

ricardoquesada, 2018-11-09:
thanks @flannelhead !

blasco-burguillos, 2019-05-25:
I have guile 2.2 and using the following doesn't seem to fix the issue. Any idea?
CT_GDB_CROSS_EXTRA_CONFIG_ARRAY="--with-guile=guile-2.2"

DasVoelkel, 2019-12-19:
> @ricardoquesada : you seem to be not using `esp-open-sdk` but building directly from `crosstool-ng` by Espressif's instructions. However, if you're getting the build failure, you can append the configuration line
> 
> ```
> CT_GDB_CROSS_EXTRA_CONFIG_ARRAY="--with-guile=guile-2.0"
> ```
> 
> to `crosstool-ng`'s `config` file (located in the same folder as the `ct-ng` binary). Do this after running
> 
> ```
> ./ct-ng xtensa-esp32-elf
> ```
> 
> but before
> 
> ```
> ./ct-ng build
> ```

how can i make this workaround work if i need the 106 compiler and not the 32 one ?

ketograph, 2020-01-02:
> I have guile 2.2 and using the following doesn't seem to fix the issue. Any idea?
> CT_GDB_CROSS_EXTRA_CONFIG_ARRAY="--with-guile=guile-2.2"

For me (using Arch Linux) using guile-2.2 didn't work either. I just installed guile-2.0 and used this.

hcet14, 2020-12-18:
I had the same error compiling with Ubuntu 20.04. Second fix from flannelhead (CT_GDB_CROSS_EXTRA_CONFIG_ARRAY="--with-guile=guile-2.0") did't work.

The first fix did the trick (CT_DEBUG_gdb=n)