pfalcon/esp-open-sdk #255: Regarding system and RF parameters on esp-open-sdk

marduino, 2017-04-09:
Seems esp-open-sdk does not load such parameters equal as ESP8266_RTOS_SDK, 

In "esp-open-sdk/common.mk", the "bootloader/firmware_prebuilt/blank_config.bin" is hardcoded to flash to 0x1000, while the RF parameter "esp_init_data_default.bin" is not included.

According with documents from ESP8266 "2A-ESP8266__IOT_SDK_User_Manual__EN_v1.3.0.pdf", in section 6.1 mentioned that:
**blank.bin** shall be programmed  to the second section from the end of flash
**esp_init_data_default.bin** stores default RF parameter values and to be written to the forth sector from the end of ﬂash

The physical address is different according with flash size.

I am very curious that why does esp-open-sdk has this different settings? And the RF parameter is ignored? Does anyone know  that?
@pfalcon

marduino, 2017-04-12:
Can anyone please response? Thanks!

someburner, 2017-04-12:
Things have changed since sdk v1.3. The SDK attempts to load the RF calibration data from a sector of flash. Supposedly that should be done automatically if you flash it to the correct sector, but I've never been able to get it to work like that with esp-open-sdk, at least not with my project. The other thing is that init data default has configuration info that may not work for every esp, or you may want to change some of the things.

What I do: 

* Add [this](https://github.com/pfalcon/esp-open-sdk/pull/234) PR
* Add `-Wl,--wrap=user_rf_cal_sector_set` to your link flags somewhere
* Then put in user main:

```
uint32 TEXT_SECTION_ATTR __wrap_user_rf_cal_sector_set(void)
{
   return 1024 - 5; // sector you have rf cal data
}
```

Also, for init_data_default, I copied the relevant parts from [nodemcu](https://github.com/nodemcu/nodemcu-firmware/blob/master/app/user/user_main.c). This injects the init_data_default into a known part of the program binary.

marduino, 2017-04-17:
em... interesting, so the function "__wrap_user_rf_cal_sector_set" is going to be invoked by some program? Do you know which program will load __wrap_user_rf_cal_sector_set? I am very interested in how do you find this function, I tried to search about this on internet for long.

Thanks very much @someburner .