pfalcon/esp-open-sdk #257: Speeding Up Build

aogriffiths, 2017-04-26:
Hi

I just raised an issue in the crosstool-ng tracker but it may be a esp-open-sdk issue. In short the build process is very slow and I'd appreciate some advice speeding it up. See below.

By way of some background - I'm dockerizing the esp-open-sdk which I think will really help people setup consistent esp8266 build environments on any machine. I plan to open source it when I have it working so will be giving it back to the community.

Many thanks
Adam

------
From https://github.com/crosstool-ng/crosstool-ng/issues/701

I'm building a toolchain into a docker images and need to regularly rebuild it; the trouble is it takes ages! I don't have the fastest internet connection but even after the tarballs have been downloaded the installing steps are very slow. The "Installing final gcc compiler" step can take well over an hour.

Can anyone give me any tips for speeding things up? For example:

Is there a way to download all the tarbars in advance and not need to re-download them every time I make the toolchain?
Can I reliably do incremental builds, i.e. not have to make clean every time?
Any other tips?
Thanks
Adam

davydnorris, 2017-04-27:
Hey Adam,

Let's do this in the actual forums under the Compiler Chat section - there are a few people like @jcmvbkbc @pfalcon @clohr @CHERTS @scargill @nkolban that would have some great advice and should be involved

itsonlym3, 2017-08-09:
sorry, but where is the Compiler Chat section (URL??)

jcmvbkbc, 2017-08-09:
http://www.esp8266.com/viewforum.php?f=9