pfalcon/esp-open-sdk #259: won't compiled on mac OS X Sierra

Grousee, 2017-05-09:
Hi, I am trying to compile the sdk on my mas OS X Sierra but i have some errors. I followed this website for installing [http://www.esp8266.com/wiki/doku.php?id=setup-osx-compiler-esp8266](url). But it failed at the end. 

[SDK_made_error.txt](https://github.com/pfalcon/esp-open-sdk/files/985795/SDK_made_error.txt)

jcmvbkbc, 2017-05-09:
Please follow [these](https://github.com/pfalcon/esp-open-sdk#macos) steps, they specifically add GNU sed to PATH.

Grousee, 2017-05-10:
I did it many times but i still have those error 

```
Makefile:181: warning: overriding commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:176: warning: ignoring old commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
make -C crosstool-NG -f ../Makefile _ct-ng
../Makefile:181: warning: overriding commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:176: warning: ignoring old commands for target `ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
./bootstrap
Running autoconf...
Done. You may now run:
    ./configure
./configure --prefix=`pwd`
checking build system type... x86_64-apple-darwin16.5.0
checking host system type... x86_64-apple-darwin16.5.0
checking for a BSD-compatible install... /usr/local/bin/ginstall -c
checking for grep that handles long lines and -e... ./configure: line 2584: 46450 Abort trap: 6           "$ac_path_GREP" -e 'GREP$' -e '-(cannot match)-' < "conftest.nl" > "conftest.out" 2> /dev/null
/usr/bin/grep
checking for egrep... /usr/bin/grep -E
checking for a sed that does not truncate output... /usr/bin/sed
checking whether sed understands -r -i -e... no
configure: error: 
make[1]: *** [_ct-ng] Error 1
make: *** [crosstool-NG/ct-ng] Error 2
```

muellermartin, 2017-05-29:
@Grousee Do you have `gnu-sed` installed via [Homebrew](https://brew.sh) and explicitly included it in `PATH` before compiling (like described in the [README under MacOS](https://github.com/pfalcon/esp-open-sdk#macos))? Because for me it works this way!

raimue, 2017-08-03:
See also #284 which will remove the need for PATH modifications.