pfalcon/esp-open-sdk #265: Migrate to upstream crosstool-ng with copied samples, overlays, local patches

davydnorris, 2017-05-23:
Will put this in now as there seems to be a push with open-lwip as well...

For almost a year now I've been using the top level crosstool-ng project to build esp-open-sdk with perfect results. I have simply renamed the submodule in esp-open-sdk, moved in the latest master from crosstool-ng/crosstool-ng, and copied the latest overlays, samples, and local patches from @jcmvbkbc into the appropriate places.

I've also been able to update gcc to 6.1.0 by copying the local patches and making some minor changes, and also the latest binutils. This has finally got LTO working for my ESP builds!

I have been talking with @jcmvbkbc and @stilor about this and I think there is general consensus between them that it makes sense to merge the work into upstream. @stilor has also talked about adding capability to fetch overlays from a URL, and @jcmvbkbc has talked about merging the newlib local patches, or doing something that's simpler to future proof.

@stilor also looked at the way we add custom configuration information to crosstool-ng and there's a formal and official way to do this with mergeconfig.sh, rather than cat and sed. I have validated this approach and it's very neat. At the moment the mergeconfig script isn't part of the kconfig suite used by crosstool-ng, but @stilor and I have discussed how it could be added as a ct-ng pass through command line option.

So my current proposal is to:
 - move the crosstool-ng submodule to the core crosstool-ng project
 - bundle the samples, overlays, and local patches from @jcmvbkbc into a tar or zip archive, along with mergeconfig.sh
 - add a one liner to the bootstrap rule in the Makefile to unzip this over the top of crosstool-ng
 - add a one liner to the clean rule to unwind this
 - change the cat and sed recipe lines in the crosstool rule to use mergeconfig.sh

I've attached a zip of the files I copy myself. This includes the local patches for gcc-6.1.0 and newlib, as well as mergeconfig.sh, to give you an idea

[crosstool-NG.zip](https://github.com/pfalcon/esp-open-sdk/files/1020606/crosstool-NG.zip)

pfalcon, 2017-05-28:
> For almost a year now I've been using the top level crosstool-ng project to build esp-open-sdk with perfect results. 

Cool, thanks for trying that, and reporting results! I'd say that many people would want to do the same.

>  This has finally got LTO working for my ESP builds!

"Working for my builds" is quite different from "working for everyone's builds in the world".

> I have been talking with @jcmvbkbc and @stilor about this and I think there is general consensus between them that it makes sense to merge the work into upstream.

I motivated @jcmvbkbc to push any patches upstream as his time permits, and glad to see a progress and momentum forming on that.

> So my current proposal is to:

Thanks, but esp-open-sdk is a simple Makefile wrapper, but with a big aspiration to become a completely open SDK for ESP8266 one day. A toolchain is very small part of that effort. Well, it's not small part, it's very important part, which has been covered by very knowledgeable people like @jcmvbkbc long ago. Beyond that long-good toolchain coverage, esp-open-sdk is conservative in respect to toolchain, so if for your ESP8266 system you get an unexpected results, which you didn't see couple of years ago, you may know that it could be your app, another vendor SDK version, another board, another AP, a dozen of other things - but NOT another toolchain, because the toolchain used is the same as used couple of years ago, well proven over this time, while other parts remain pretty flaky, so stability at least in the toolchain is a must.

So, I appreciate your effort and encourage you to continue, but esp-open-sdk has more important things in agenda than toolchain which works pretty well. More info: https://github.com/pfalcon/esp-open-sdk/pull/264#issuecomment-304497781

closed 2017-05-28 by pfalcon

davydnorris, 2017-05-29:
ok, well I'll continue to do my own thing - no worries. Would you at least consider my fix for your broken crosstool-config-overrides? That would let people who are interested in more recent toolchain components build them without having to hack crosstool samples files.

If you are then it's one file addition and a couple of lines changed in the Makefile.

pfalcon, 2017-05-29:
> Would you at least consider my fix for your broken crosstool-config-overrides?

Sure, feel free to submit a patch with arguments why yours is less broken than "mine" (git log crosstool-config-overrides), and it will be processed with [If it ain't broke, don't fix it](https://en.wikipedia.org/wiki/Wikipedia:If_it_ain%27t_broke,_don%27t_fix_it) priority.