pfalcon/esp-open-sdk #266: Faild to build on Windows Subsystem for Linux in Windows 10

cspoonerKRHS, 2017-06-03:
I'm getting a build failure in Bash on Ubuntu on Windows. Here is the error message:

`spooner@SpoonXPS:~/dev/esp-open-sdk$ make
Makefile:187: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:178: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
cp esptool/esptool.py /home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/bin/
ln -snf ESP8266_NONOS_SDK_V2.0.0_16_08_10 sdk
Installing vendor SDK headers into toolchain sysroot
Installing vendor SDK libs into toolchain sysroot
Installing vendor SDK linker scripts into toolchain sysroot
make -C lx106-hal -f ../Makefile _libhal
make[1]: Entering directory '/home/spooner/dev/esp-open-sdk/lx106-hal'
../Makefile:187: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:178: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
autoreconf -i
PATH=/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Python27:/mnt/c/Python27/Scripts:/mnt/c/Program Files (x86)/Intel/iCLS Client:/mnt/c/Program Files/Intel/iCLS Client:/mnt/c/Windows/System32:/mnt/c/Windows:/mnt/c/Windows/System32/wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Windows/System32:/mnt/c/Windows:/mnt/c/Windows/System32/wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0:/mnt/c/Program Files (x86)/GtkSharp/2.12/bin:/mnt/c/Users/Spooner/AppData/Local/Microsoft/WindowsApps ./configure --host=xtensa-lx106-elf --prefix=/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr
/bin/bash: -c: line 0: syntax error near unexpected token `('
/bin/bash: -c: line 0: `PATH=/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/mnt/c/Python27:/mnt/c/Python27/Scripts:/mnt/c/Program Files (x86)/Intel/iCLS Client:/mnt/c/Program Files/Intel/iCLS Client:/mnt/c/Windows/System32:/mnt/c/Windows:/mnt/c/Windows/System32/wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/DAL:/mnt/c/Program Files (x86)/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files/Intel/Intel(R) Management Engine Components/IPT:/mnt/c/Program Files (x86)/NVIDIA Corporation/PhysX/Common:/mnt/c/Windows/System32:/mnt/c/Windows:/mnt/c/Windows/System32/wbem:/mnt/c/Windows/System32/WindowsPowerShell/v1.0:/mnt/c/Program Files (x86)/GtkSharp/2.12/bin:/mnt/c/Users/Spooner/AppData/Local/Microsoft/WindowsApps ./configure --host=xtensa-lx106-elf --prefix=/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr'
../Makefile:167: recipe for target '_libhal' failed
make[1]: *** [_libhal] Error 1
make[1]: Leaving directory '/home/spooner/dev/esp-open-sdk/lx106-hal'
Makefile:164: recipe for target '/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libhal.a' failed
make: *** [/home/spooner/dev/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/libhal.a] Error 2`

I've run apt-get update && apt-get upgrade; and all of the recommended packages installed without issue.

pfalcon, 2017-06-03:
On real unix, paths never contain funny characters like " " or "(". They may - but only for people who know how to deal with that. Consequently, you either don't have funny paths in your PATH, or know how to deal with that, choose solution which is easier for you (most people choose 1st one).

closed 2017-06-03 by pfalcon

cspoonerKRHS, 2017-06-03:
Thanks for catching that...I just moved from Linux to windows 10 (new computer) and missed that. I'll have to see if I can fix that.

someburner, 2017-06-04:
>I just moved from Linux to windows 10

:open_mouth:

profra, 2017-06-04:
Try this in your script: ... it helped me... eventhough I had "funy spaces" in the windows path 
```
..
export PATH=/home/fp/github/esp-open-sdk/xtensa-lx106-elf/bin:"$PATH"
..
```
the only change is to put variable $PATH into "$PATH"..