pfalcon/esp-open-sdk #267: Make fails on WSL while making lx106-hal

skbobade, 2017-06-04:
WSL runs on Ubuntu 16.04 which has Perl version is 2.22.1. Perl has deprecated support for braces without escape character.
This error happens when during make PATH is being appended to include xtensa.
Windows appends path of Bash with Windows program directories to make it possible to run Windows application from Linux layers. Thus some of the directories that are already in PATH contains text "(x86)"

This is okay in an older version of Perl, results in an error during make with this setup. However, I ran `make` multiple times without `make clean` after a few retries installation was successful.

Not sure this problem is in error/warning handling in .configuration, automake or make file or esp-open-sdk or problem with lx106-hal configurations.

pfalcon, 2017-06-04:
If you're sure it's lx106-hal problem, please report it to it (which is a separate project). Otherwise, report it to WSL.

skbobade, 2017-06-04:
@pfalcon I'm not sure! Reported to lx106-hal already. Why do you think this issue should be reported to WSL?

pfalcon, 2017-06-04:
Please add link to lx106-hal ticket.

If the issue happens on WSL and doesn't happen on normal Unix system, then practically, WSL is the culprit.

profra, 2017-06-04:
Try this in your script: ... it helped me... eventhough I had "funy spaces" in the windows path

..
export PATH=/home/fp/github/esp-open-sdk/xtensa-lx106-elf/bin:"$PATH"
..
the only change is to put variable $PATH into "$PATH"..

skbobade, 2017-06-04:
@pfalcon Here is lx106-hal ticket. https://github.com/tommie/lx106-hal/issues/12

skbobade, 2017-06-04:
@profra I did this for exporting xtensa path. However, I could not understand where in scripts PATH is being passed without quotes.

And I don't understand how simply executing `make` multiple times finally gets the script going (I think 3rd time was the charm)

Build log for error attached if interested.

[build.txt](https://github.com/pfalcon/esp-open-sdk/files/1050269/build.txt)

tommie, 2017-06-04:
I think this is due to unescaped paths in https://github.com/pfalcon/esp-open-sdk/blob/b069537944fab5d3a260b82a9aaaad5c560dd6bd/Makefile#L168.

skbobade, 2017-06-07:
Pull request #218 solves the issue.

pfalcon, 2017-06-23:
@skbobade : Thanks for testing. Merged.

closed 2017-06-23 by pfalcon