pfalcon/esp-open-sdk #268: Would the project consider adding mbedTLS as well?

davydnorris, 2017-06-07:
Given the existing SDK SSL libraries are pretty broken and Espressif is encouraging the move to TLS (and has released their own), would it be something this project would consider?

I notice the esp-open-rtos folks have lwip and mbedTLS as part of their dist.

LongHairedHacker, 2017-06-07:
I've been looking into this myself for the last few evenings, trying to figure out how the esp-open-rtos folks actually got it to work.

So looking at https://github.com/SuperHouse/esp-open-rtos/blob/master/extras/mbedtls/net_lwip.c I'd say they use the sequential socket API provided by lwip to implement a what mbedtls calls a TCP/IP communication module (see https://tls.mbed.org/api/group__tcpip__communication__module.html).
In order to provide the sequential API lwip needs a few support functions like message passing and semaphores, that are used to turn the callback based API into a sequential one.
esp-open-rtos implements those using parts of FreeRTOS. 
See https://github.com/SuperHouse/esp-open-rtos/blob/master/lwip/sys_arch.c
Implementing the support functions without FreeRTOS will be difficult.

At least that's my understanding of the issue.
I'd be interested to read a second opinion for my findings.

Maybe it is possible to work around this issue similarly to how the Arduino folks integrated axTLS.

davydnorris, 2017-06-08:
I've just compiled the mbedTLS demo that comes in the Espressif supplied zip - it comes with with the mbedTLS library source code as well:

[http://espressif.com/sites/default/files/sdks/esp8266_nonos_sdk_mbedtls_20160718.zip](http://espressif.com/sites/default/files/sdks/esp8266_nonos_sdk_mbedtls_20160718.zip)

It appears the code is pretty much untouched, but I haven't checked the version. They have configured mbedTLS using the standard mechanisms built into the library, defining the usual memory management and formatted string printing routines etc. and using lwIP functions where possible. They have then added a layer to implement mapping of some lwIP functions to the underlying ROM, and have also added implementations of the overlying esp_secure_* functions so that they call mbedTLS instead of SSL.

I turned on all warnings and fixed quite a few implicit declarations, finding a few missing from the SDK headers in the process. I also had to fix their code mapping of the memory routines as mbedTLS uses standard malloc, calloc and they had them mapped to pvPortCalloc etc.

davydnorris, 2017-06-08:
Just checked the code and the library that Espressif is shipping is mbedTLS 2.2.1.

I'm going to have a bash at pulling in the latest release and see how it goes

pfalcon, 2017-06-08:
> would it be something this project would consider?

No. Users of this project can use *any* of available SSL/TLS libraries, or any of millions existing libraries at all, and there's no need to bring them into this project. The scope of this project is trying to replace vendor's binary blobs directly related to esp8266 with open sources. That alone are man-years of work, to not try to pick up unrelated tasks.

closed 2017-06-08 by pfalcon

davydnorris, 2017-06-08:
LOL

No worries mate. You're a cheery bloke aren't you.

pfalcon, 2017-06-08:
@davydnorris : I definitely am. Have a look at https://github.com/pfalcon vs https://github.com/davydnorris, compare number of repos, number and frequency of commits, research or take my word that for me, it's like that last 3 years. And you tell me that you want to do more, more, and more? Let me not believe and tell you that everyone wants to do less, definitely not more than really necessary.

davydnorris, 2017-06-08:
Yeah I never use Git - I'm not a huge Git fan TBH so I avoid it if possible. I have used enterprise level SCM  for the past 25 years and contributed code to projects for the past 35. If you searched you might find some of my bits and pieces still lying around, but none of that's very important :-) 

I do love to code and to help get the most out of little boxes, and I do appreciate what you are doing here, but I know the later versions of the tools and libs are offering a lot, and a lot that I need for my work.

I understand your position, and this is your repo so you are entitled to do with it whatever you want. It was an amazing leg up for me when I first started with this chipset, and for that I offer huge thanks. But I guess you just now made me realise I have reached the stage where I've: 
 - gotten the tools building on an uncommon platform for them (Cygwin64) and contributed fixes
 - been replacing crosstool with the latest HEAD revisions and building for nearly 9 months
 - replaced the official 4.8.5 version of the compiler and tools with the 6.x series and added LTO
 - started working on porting later versions of Newlib
 - almost completely rewritten the Makefile
 - replaced SSL with mbedTLS
It's kind of like my Dad's joke about the family axe being handed down for 150 years - "it's only had 3 new heads and 20 new handles...." :-)

So I'm going to stop asking. And I am going to contribute only what might be seen as bug fixes, although you have done a great job keeping this stable. And like CHERTS and a bunch of other users, I'll thank you for my learning experience from this and roll my own. Evolution and divergence is a good thing!

I have one 'bug fix' in the pipeline for correct handling of config changes in ct-ng where the Makefile uses a mix of sed and the crosstool-config-overrides file, but have been super busy and so haven't been able to pull just that section out of my Makefile and diff it. I hope it will be of use. Credit to stilor for suggesting the approach

pfalcon, 2017-06-08:
If you understand my position, there's little to argue about.

> this is your repo so you are entitled to do with it whatever you want

Not what I want, but what I realistically can (also interested in). If a maintainer accepts somebody's patches, they agree to maintain and support them over time. If they not ready to do that, I consider it's fair to tell that to a (potential) contributor.

I tried to explain the approach towards toolchain, and also tried to hint about approach to mbedTLS. If you need more hints, README refers to MicroPython as a real-world project using esp-open-sdk. If you look at it, you'll find: a) it's a portable project, esp826 being just one of many platforms it supports; b) it provides mbedTLS integration (of some level), completely not tied into esp8266 or any other platform. That's how I do it, and I'm not interested in shipping mbedTLS specifically for esp8266, I'm not an Espressif engineer to care about esp8266 *that* much.

Finally, I referred you to esp8266 RE project, https://github.com/pfalcon/xtensa-subjects/tree/master/2.0.0-p20160809 as means to apply effort in the direction I consider "right". I may imagine you aren't interested in that, but then it shouldn't come as surprise that your project look that way to other people too.

You're welcome to use tickets in this project to discuss things with other interested parties of course (as long as there're such parties), and this ticket can be reopened if that helps.