pfalcon/esp-open-sdk #279: Blinky example don't start on esp12f

de1m, 2017-07-07:
Hi,
I've installed the esp-open-sdk on my linux.
The previously version that I've used was the SDK V1.5 and with this version working my esp12 without any problem. 
Now for a new project I've installed the V2.0.0_16_08_10 and compiled for a test the same blinky example (this has worked on sdk 1.5). But now my esp12 don't working(after compiling and flash).
For my test I've used the Makefile from blinky example test.
```
de1m@comp0 /opt/esp-open-sdk/examples/blinky $ esptool.py -p /dev/ttyUSB0 write_flash --flash_mode dio 0 blinky-0x00000.bin 0x40000 blinky-0x10000.bin
esptool.py v1.2
Connecting...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0240
Writing 36864 @ 0x0... 36864 (100 %)
Wrote 36864 bytes at 0x0 in 3.3 seconds (90.6 kbit/s)...
Writing 196608 @ 0x40000... 196608 (100 %)
Wrote 196608 bytes at 0x40000 in 17.3 seconds (90.9 kbit/s)...
Leaving...
```

soocki, 2017-07-10:
Similar problem on ESP8266 (Wemos D1 mini Pro). It worked couple days back, now the built-in LED just blinks rapidle after flashing. No errors, compiles and flashes but doesn't work. Not sure how to proceed.

soocki, 2017-07-13:
There's some working examples available in [esp-open-rtos](https://github.com/SuperHouse/esp-open-rtos), it worked for me.

pfalcon, 2017-07-13:
Everything known to work. 1) Check your hardware. 2) Check https://www.chiark.greenend.org.uk/~sgtatham/bugs.html

de1m, 2017-07-13:
I think hardware is okay, yesterday I've installed Arduino for ESP and it's have working without any problem. Also my last installation with sdk 1.5.4 has working. 

So it's not a hardware error. Wich info do you need?

d-a-v, 2017-07-13:
I am also switching from arduino to esp-open-sdk and back, and experienced similar behaviours.
The solution that worked for me is to completely blank the flash using the different available versions of esptool (there are two in esp-open-sdk, and another one in arduino).

de1m, 2017-07-13:
I've installed yesterday my new pc and also a fresh esp-open-sdk and esptool from python pip (version 2.1).
I've erased flash with old version (from esp-open-sdk v1.2) and with the new version of esptool 2.1, then I've flash the blinky example first with esptool 1.2 then with 2.1.
Also I've used manuall the old address of memory (0x0000 and 0x4000) but always the same behaviours
```
esptool.py -p /dev/ttyUSB0 write_flash --flash_mode dio 0 blinky-0x00000.bin 0x40000 blinky-0x10000.bin
esptool.py -p /dev/ttyUSB0 write_flash --flash_mode dio 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin
``` 
Ich habe keine idee mehr, was kann ich tun
Ich glaube, ich mache ein Downgrade auf sdk 1.5.4

d-a-v, 2017-07-13:
At one point I made a blank file (dd if=/dev/zero) and flashed it, then it worked after a regular flash (all esptool "blank" options were not working). I was indeed suspecting an hardware fault before that.
You could try that, and trying back the 1.5.4 sdk too if it fails.

de1m, 2017-07-13:
okay, thank you for info. I try this today evening

de1m, 2017-07-13:
It's not working. So my steps
1. Create a dummy file with dd
2. flash this file on esp12f (esptool 1.2)
3. flash blinky example 
```
de1m@comp0 /opt/esp-open-sdk-a/examples/blinky $ make flash
esptool.py write_flash 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin
esptool.py v1.2
Connecting...
Auto-detected Flash size: 32m
Running Cesanta flasher stub...
Flash params set to 0x0040
Writing 36864 @ 0x0... 36864 (100 %)
Wrote 36864 bytes at 0x0 in 3.3 seconds (90.7 kbit/s)...
Writing 196608 @ 0x10000... 196608 (100 %)
Wrote 196608 bytes at 0x10000 in 17.3 seconds (90.9 kbit/s)...
Leaving...
```
now the chip blink with a very high frequency (it's not a correct frequency 500ms)

d-a-v, 2017-07-15:
have you tried going back to arduino environment with this chip ?
there's this arduino sketch [https://github.com/kentaylor/EraseEsp8266Flash](this) to try before trying esp-open-sdk/blinky again.

de1m, 2017-07-16:
yesterday I' received a new dev board modul (http://www.ebay.de/itm/NodeMCU-ESP8266-ESP-12E-V1-0-Wifi-CP2102-IoT-Lua-267-/172778652438?hash=item283a690316)
I've make steps like above
1.Create a dummy file with dd
2.flash this file on esp12f (esptool 1.2)
3. flash blinky example
...but the save behaviour (the chip blink with a very high frequency (it's not a correct frequency 500ms))

d-a-v, 2017-07-16:
Sorry I still have little experience with esp-open-sdk. I observe the same behaviour with micropython when trying lwIP-v2 and I'm still looking for time to sort this out. I'll try simple/blinky tests too when I have time. The trick above worked once.
Given that arduino sketches are working, try with the erase sketch before flashing the blinky example.
Have you tried all the "erase" options from all available esptool versions ?

tarakanov, 2017-07-18:
I also experience this problem. I have ESP12 module. Hardware is OK, blink example from  esp-open-rtos is working (with the same sdk), but blink example from SDK result in rapidly flashing LED

de1m, 2017-07-18:
@tarakanov what OS do you use? Windows? Linux?

Navis-Raven, 2017-07-18:
I have exactly the same problem, I use the blinky firmware available here 

The basic blinky example works WELL with an ESP-12E but NOT with an ESP-12F

In ordrer to install all of esp-open-rtos toolchain and and blinky code example I've approximatively followed this protocol here at https://www.penninkhof.com/2015/03/esp8266-open-sdk/#comment-698  
When flashing with esptool.py it works very well with an ESP-12E but not with the same board and an ESP-12F instead.

Additionnal infos:
I've the same problem by using it trough eclipse.
I have exactly the same problem with two other of my firmwares. 

(I used both of Ubuntu and Debian, it didn't change anything)
This is a critical issue.

Navis-Raven, 2017-07-18:
I think the ESP-12F after flashing, does not wakes up from the boot mode since instead of having 3.3V on the GPIO 0 pin, this pin outputs an alternative signal (with an offset) which may be the quartz signal (I think that te ESP-12 outputs the quartz signal on GPIO 0 when on boot mode). Please confirm this info.

(I've observed this triangular/alternative signal with an oscilloscope, on two ESP-12F but not on the ESP-12E except on the boot mode or at the beginning after reset)

I need to confirm these infos

Navis-Raven, 2017-07-18:
"I also experience this problem. I have ESP12 module. Hardware is OK, blink example from esp-open-rtos is working (with the same sdk), but blink example from SDK result in rapidly flashing LED"
@tarakanov 

The problem is that the blinky example does not start at all, if you have seen the blinking correctly, and you certify that the firmware works but at high speed so it is just a problem of frequency and you've solved the problem. 

So, you should use an UART to be sure that your firmware is running, and it is not just a crystal noise oscillating like the one one GPIO 0.

Navis-Raven, 2017-07-18:
I don't know if this issue is the same on ESP OPEN RTOS, we'd better verify

Navis-Raven, 2017-07-18:
After verification @tarakanov  @de1m  there is no signal on the led in my case, please verify that you have an ESP-12F and not an ESP-12E or other. It seems that our issues are not really the same even, in other words it seems that the cause of both issues are not the same

Navis-Raven, 2017-07-18:
I confirm, I have exactly the same problem with two other of my firmwares and one of them is running with esp-open-rtos. 

Does esp-open-rtos uses libraries from esp-open-rtos ?

bamartin125, 2017-07-19:
@Navis-Raven  - maybe you should check to see that you are set up on the correct GPIO for your model? Maybe you have already confirmed that...

@tarakanov  - I had the same problem as you it seems. I found a post here that helped: [https://blog.vinczejanos.info/2016/11/14/unbrick-esp8266-blinking-blue-led/](https://blog.vinczejanos.info/2016/11/14/unbrick-esp8266-blinking-blue-led/)

Essentially, I ended up flashing some of the binary blobs from esp-open-sdk to my ESP using esptool.py. First though (as the post suggests), I found out what model SPI flash I had by issuing `esptool.py --port /dev/ttyUSB0 flash_id` and used the suggested file here [https://code.coreboot.org/svn/flashrom/trunk/flashchips.h](https://code.coreboot.org/svn/flashrom/trunk/flashchips.h) to figure out what my flash chip size was. I used the table in the post in order to determine what addresses I should flash the blobs to since I found my flash size to be 4MB.

I could not find the master_device_key.bin file, so I only did the following (please remember to check the table for the appropriate addresses for your model flash memory):

1. `cd esp-open-sdk/ESP8266_NONOS_SDK_V2.0.0_16_08_10/bin`
2. `esptool.py write_flash 0x3FC000 esp_init_data_default.bin`
3. `esptool.py write_flash 0x3FE000 blank.bin`
4. `cd at_sdio/noboot`
5. `esptool.py write_flash 0x00000 eagle.flash.bin`
6. `esptool.py write_flash 0x40000 eagle.irom0text.bin`

I then went back and ran `make flash` for the blinky example and that worked for me finally. Does someone know which of those flashes was necessary for things to start working? Maybe everything was?

jeremyd2019, 2017-07-19:
@bamartin125 - I think it's this one
> `esptool.py write_flash 0x3FC000 esp_init_data_default.bin`

I've been following this thread, with the vague recollection of a new blob added in SDK 2.x that needed to be flashed before things built with SDK 2.x would run.  It's been quite a while since I've looked at this stuff, so I couldn't remember the details.  This has jogged my memory, esp_init_data_default.bin was the one added in SDK 2.x.

Navis-Raven, 2017-07-19:
@tarakanov @jeremyd2019 
So in facts our issues are quite different. But if your led blinks, it might signify that your programm is running, if it is the case it is really not the same issue : please verify with an UART and printf("hello_world\n");
If your programm displays hello_world in your Putty terminal (or equivalent) so your Blinky programm is running. So in this case the issue is not the one in the titile "Blinky example don't start on esp12f". If not already tryied, try to modify the blinky programm in order  to decrease the frequency of led blinking (equivalent to increase the temporisation between each blinks)

If you're sure that your programm does not run, please verify the signal on the GPIO 0, and the one on the ESP's LED and describe it. 

We have to be sure that we are talking about the same problems. Because it seems that there is many issues in one here.

Navis-Raven, 2017-07-19:
@EVERYBODY:  
We have to be sure that we are talking about the same bug:
1) Please verify the version of the ESP-12 you use, and be sure that this is the ESP12-F (as said in the topic title) and not the ESP-12E. 
2) Verify that your blinky programm does not start, (try to send a hello world message in the UART
3) Check the signal at GPIO 0 with an oscilloscope, and describe it.
4) Check your SDK version.
5) Does your same Blinky programm with the same PC configuration (compilation flashing toolchain etc...) works on an ESP-12E.
It is a very important step to investigate this bug.

Navis-Raven, 2017-07-19:
If @tarakanov and @jeremyd2019  have only a problem of frequency with their ESP-12F (if it is an F and not an E), we can apporach more a solution.

Navis-Raven, 2017-07-19:
For me,

1. I use the ESP-12F
2. The blinky program doesn't start at all. No hello world is printed into the putty console. 
3. The voltage GPIO 0 pin is a triangular/alternative  signal (with an offset) instead of a 3.3V
4. I use the latest SDK version.
5. The same basic and standard blinky programm works on ESP-12E, and not on the ESP-12F. 
6. I've tried on Ubuntu and Debian

Navis-Raven, 2017-07-19:
@de1m 
Your component (link below) IS NOT an ESP-12F but an ESP-12E. In conclusion, your problem is not at all the same of this topic, you shall open a new issue. 
"NodeMCU ESP8266 ESP-12E V1.0 Wifi CP2102 IoT Lua 267"

http://www.ebay.de/itm/NodeMCU-ESP8266-ESP-12E-V1-0-Wifi-CP2102-IoT-Lua-267-/172778652438?hash=item283a690316

 If not already tryied, try to modify the blinky programm in order to decrease the frequency of led blinking (equivalent to increase the temporisation between each blinks)

de1m, 2017-07-19:
@Navis-Raven yes I can,
1. Create a file in Linux with "dd" - d if=/dev/zero of=blank.bin bs=4096K count=1
2. Write this file to esp - esptool --port=/dev/ttyUSB0 write_flash 0 blank.bin

I've changed the timer settings in blinky example, but frequency is the same
```
  // setup timer (500ms, repeating)
  os_timer_setfn(&some_timer, (os_timer_func_t *)some_timerfunc, NULL);
  os_timer_arm(&some_timer, 1100, 1);
```

Navis-Raven, 2017-07-20:
Please careful about this issue https://github.com/pfalcon/esp-open-sdk/issues/226, the Blinky makefile is obsolete and not up to date !

Navis-Raven, 2017-07-20:
I have some news.
After many trials I found something, 
I've seen that doing this on an ESP-12E
esptool.py --port=/dev/ttyUSB0 write_flash 0x00000 blank.bin
has the same results as flashing the ESP-12F with the basic blinky programm or any other programm. 

So I've suspected that it was a flash problem, I've tried many flash modes, and this is what I found:

The ESP-12F I have, in the countrary of the ESP-12E model I have must be programmed in "dout" mode. But I really don't now why. 
This is the command line I did: 
esptool.py -p /dev/ttyUSB0 write_flash --flash_mode dout 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

This are the results, I have three different hardwares (HW1, HW2, HW3) with ESP-12E and ESP-12F:

HW1+ESP-12E : Working with high speed blinking
HW1+ESP-12F : Working normally (blinks every 500ms)
HW2+ESP-12E : Working with high speed blinking
HW3+ESP-12F : Working with high speed blinking 

We can conclude at least that we are NOT facing the same issue as the beginning. These are two different issues. 

The most weird thing is that after making only one "--flash_mode dout" on HW2+ESP12F, the next times I did not event had to add this flash mode dout option to succeed into flashing, however it was necessary on the HW3+ESP-12F. Very weird !

Navis-Raven, 2017-07-20:
It seems that the high speed blinking problem is a high speed rebooting problem. 

@tarakanov @de1m  you have to do an UART with a putty condole then you will observe it.

Navis-Raven, 2017-07-21:
I've did all my tests above with the software availablehere: https://github.com/esp8266/source-code-examples/tree/master/blinky
I remade it with the software
https://github.com/pfalcon/esp-open-sdk/tree/master/examples/blinky

No notable change remarked

zn4321, 2017-07-23:
I've run into same problem today. I seems that first time should download both the esp_init_data_default.bin and blank.bin files. So least once:

esptool.py write_flash 0x00000 blinky-0x00000.bin 0x10000 blinky-0x10000.bin 0x7C000 esp_init_data_default.bin 0x7E000 blank.bin

once you did, the "make flash" will works too.

Navis-Raven, 2017-07-25:
@zn4321  Unfortunately it gaves me no results at this time

Navis-Raven, 2017-07-25:
@zn4321  with which ESP-12 did you work ?

zn4321, 2017-07-25:
My module is an ESP-100 module but the symptoms was same. One more thing is that the esptool.py (version 1.2) provided by the SDK is not work for me. But the version 2.1-beta ([from github](https://github.com/espressif/esptool)) is works fine.

Navis-Raven, 2017-07-26:
ESP-100 ? I did not find any information about this module ? Can you give me some links ?

zn4321, 2017-07-26:
[here](https://www.banggood.com/ESP8266-ESP-100-Serial-WIFI-Module-Compatible-With-ESP-07-External-Antenna-p-1055445.html) or [here](http://www.ebay.com/itm/ESP8266-ESP-100-Wifi-Module-Serial-Compatible-for-ESP-07-/262591881011) etc. This is  modified version of ESP-07, it have 8MB flash memory. The 1.2 version of esptool.py can't recognize the flash type which used on this ESP-100.

jeremyd2019, 2017-07-26:
@zn4321 According to [this](https://github.com/espressif/ESP8266_NONOS_SDK/commit/90c641efe84066b47c4616ed367697a9f49f3ac5), support for 8MB and 16MB flash was added to the ESP8266_NONOS_SDK in version 2.1.0.  Also you'd probably need to adjust the addresses on the esptool invocation.  Hopefully they updated the flash map section in their documentation when they added support for the higher capacity flash chips.

Navis-Raven, 2017-07-27:
So it might be possible that it is a bug of esptool.py

Navis-Raven, 2017-07-27:
I've solved one of the bugs I had. 
I had a personnal blink software with esp-open-rtos and I make it work in QIO, DIO, QOUT and DOUT with all hardware ESP-12E and ESP-12F except one, one of my hardware which has an ESP-12F was working in all flash modes DIO, QOUT, DOUT but not in QIO. I verified with esptool.py flash_id command that the two ESP-12F had the same flash memory and both of them has the same gigadevice GD25Q32 which is QIO compatible as said here https://www.elnec.com/en/device/GigaDevice+Semic./GD25Q32/
So the ESP-12F is QIO compatible !

So I think that one of pin of the GD25Q32 flash memory inside the ESP-12F module is disconnected or not correctly soldered. This is a hardware problem INSIDE one of my two ESP-12F. So I can run it only on QOUT/DOUT/DIO.

See this topic for more infos about these QIO/DIO/DOUT/QOUT problem
see this topic https://www.esp32.com/viewtopic.php?t=1250

Navis-Raven, 2017-07-27:
I finally succeeded into making it work well with blinky.

This is what I did: 
I used the makefile of open SDK with the make flashinit command. 
Which is equivalent to the command:
/opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB0 write_flash -ff 40m -fm qio -fs 32m \
	0x3e000 /opt/esp-open-sdk/sdk/bin/blank.bin \
	0x3fc000  /opt/esp-open-sdk/sdk/bin/esp_init_data_default.bin \
	0x3fe000  /opt/esp-open-sdk/sdk/bin/blank.bin
And so the blank.bin and esp_init_data_default.bin are programmed at theses addresses 
blank.bin-------->0x3e000
blank.bin-------->0x3fe000 
esp_init_data_default.bin-------->0x3fc000 

(maybe those are the binaries available here: https://github.com/espressif/ESP8266_NONOS_SDK/tree/master/bin , to confirm)

After this I programmed the blinky programm with this command:
esptool.py -p /dev/ttyUSB0 --baud 115200 write_flash -fs 32m -ff 40m -fm dout 0 blinky-0x00000.bin 0x10000 blinky-0x10000.bin

And it works

Note that YOU MUST precise the flash "-fs 32m" or it would not work.
I remind that I used dout flash mode ("-fm dout"), since one of my two ESP-12F does not work in QIO mode.

Navis-Raven, 2017-08-03:
UPDATE:
If it still does not work try thiscommand (after downloading the corresponding files) : /opt/esp-open-sdk/esptool/esptool.py -p /dev/ttyUSB* -b 230400 write_flash -ff 40m -fm qio -fs 8m 0x00000 /opt/esp-open-sdk/sdk/bin/boot_v1.5*.bin

Curly-, 2017-08-06:
The high speed flashing LED means that the ESP failing to boot and is sending an error over uart. If you monitor the serial output with the bootloader's baud rate of 76800, you should see an error. If the error message contains the line `rf_cal[0] !=0x05,is 0xFF`, this means that the rf_cal data has been erased, and not restored. To restore it flash esp-open-sdk/ESP8266_NONOS_SDK_V2.0.0_16_08_10/bin/esp_init_data_default.bin to the appropriate memory address. The correct address for your ESP can be found on [this page](https://github.com/nodemcu/nodemcu-firmware/blob/master/docs/en/flash.md#sdk-init-data). You shouldn't need to write blank.bin, and the default Makefile for blinky should work fine, without any modifications.

edgardogho, 2017-10-27:
Navis-Raven you got it!.
I had the same issue with the flash size and using your method on esptool it worked perfectly.
Thanks!

ballamuth, 2017-12-26:
Actually, since SDK 1.5.2 you must add ```uint32 user_rf_cal_sector_set( void )``` function into your application to specify the location of the custom RF calibration sector (instead of default 0x3FC000 for 32M flashes, or other addresses for flashes of different size). And if you look into 0x10000.bin image, you'll find that data from esp_init_data_default.bin is located there at offset 0x2D810. Hope that helps (=

DeuxVis, 2018-03-02:
Many thanks @Navis-Raven !

I had the same symptoms on a wemos D1 : very fast flashing led resulting, whatever the code I programmed using esp-open-sdk is actually supposed to do, BUT programming anything in arduino IDE working fine.

I used the suggestion in your last 27 Jul 2017 post above, and now I finally have that blinky example working fine.