pfalcon/esp-open-sdk #280: BBS is down

russell, 2017-07-20:
The BBS is down because it can't connect to it's own MySQL server.  Can these files be included in the repo instead of relying on the attachments in the BBS?

```cp esptool/esptool.py /home/espbuilder/esp-open-sdk/xtensa-lx106-elf/bin/
Creating irom version of libc...
/home/espbuilder/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-objcopy --rename-section .text=.irom0.text \
	--rename-section .literal=.irom0.literal /home/espbuilder/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libc.a /home/espbuilder/esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/lib/libcirom.a;
wget --content-disposition "http://bbs.espressif.com/download/file.php?id=1690"
converted 'http://bbs.espressif.com/download/file.php?id=1690' (ANSI_X3.4-1968) -> 'http://bbs.espressif.com/download/file.php?id=1690' (UTF-8)
--2017-07-20 16:46:46--  http://bbs.espressif.com/download/file.php?id=1690
Resolving bbs.espressif.com (bbs.espressif.com)... 119.9.92.91
Connecting to bbs.espressif.com (bbs.espressif.com)|119.9.92.91|:80... connected.
HTTP request sent, awaiting response... 503 Service Unavailable
2017-07-20 16:46:47 ERROR 503: Service Unavailable.

Makefile:366: recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip' failed
make: *** [ESP8266_NONOS_SDK_V2.0.0_16_08_10.zip] Error 8
```

ulno, 2017-07-21:
Seems like the NONOS zips have a permissive license (ESPRSSIF MIT). Can we re-upload these to another repository here on github? Most of us will have a copy of them, so we could just re-share and change the source-path.

jeremyd2019, 2017-07-21:
I just checked and ESP8266_NONOS_SDK seems to have [a GitHub presence now](https://github.com/espressif/ESP8266_NONOS_SDK).  Maybe either open an issue on there asking them to upload older builds to the releases section, or else somebody could fork it as a starting point.

russell, 2017-07-24:
They do have some of the files hosted on their site http://espressif.com/en/products/software/esp-sdk/resource which is probably more reliable than the BBS.  But if they are moving towards hosting on git anyway, this problem might go away on it's own.  I'm going to close this, I think the most productive thing would be to rely on their git tags in the future.

closed 2017-07-24 by russell