pfalcon/esp-open-sdk #285: non-standalone installation: ld cannot open linker script file eagle.app.v6.ld

DavidAntliff, 2017-08-06:
I've installed this project as per the README, using `make STANDALONE=n`, on Ubuntu 16.04. It was successful as far as I can tell. I have added the xtensa path to PATH.

I have modified the Blinky Makefile to set the include path for CFLAGS and LDFLAGS:

    CFLAGS += -I/home/me/esp-open-sdk/sdk/include
    LDFLAGS += -L/home/me/esp-open-sdk/sdk/lib

When I try to build the Blinky example, I get the following error:

```
$ make
xtensa-lx106-elf-gcc -I. -mlongcalls -I/home/me/esp-open-sdk/sdk/include   -c -o blinky.o blinky.c
xtensa-lx106-elf-gcc -Teagle.app.v6.ld -L/home/me/esp-open-sdk/sdk/lib  blinky.o  -nostdlib -Wl,--start-group -lmain -lnet80211 -lwpa -llwip -lpp -lphy -lc -Wl,--end-group -lgcc -o blinky
/home/me/esp-open-sdk/xtensa-lx106-elf/lib/gcc/xtensa-lx106-elf/4.8.5/../../../../xtensa-lx106-elf/bin/ld: cannot open linker script file eagle.app.v6.ld: No such file or directory
collect2: error: ld returned 1 exit status
<builtin>: recipe for target 'blinky' failed
make: *** [blinky] Error 1
```

The file `eagle.app.v6.ld` appears to exist in the `esp-open-sdk/sdk/ld` directory but for some reason the build system is not able to find it.

Is this an error on my part (an environment variable I forgot to set perhaps?) or an issue with this version of the project. I cloned it from master today: 350c0e9

DavidAntliff, 2017-08-06:
The build was successful if I modify the Makefile to specify the full absolute path to the .ld file:

    LDFLAGS = -T/home/me/esp-open-sdk/sdk/ld/eagle.app.v6.ld

Or if I add the `sdk/ld/` directory to the LD search path:

    LDFLAGS = -Teagle.app.v6.ld -L/home/me/esp-open-sdk/sdk/ld

Perhaps this suggests that the `*.ld` files have moved into `sdk/ld/` since Blinky's makefile was written? Perhaps they were originally in `sdk/lib/` (as the `-T` option uses the `-L` search path), so `ld` would have found them in there before they moved. I'm only speculating though.

pfalcon, 2017-11-18:
That's exactly why there's "standalone" setup and why it's default - it "just works". Please use non-defaults only if you know what you do.

closed 2017-11-18 by pfalcon

DavidAntliff, 2017-11-19:
To be fair, the README does not make this point clear. It presents "non-standalone" as a valid choice. I don't think it was unreasonable to question this aspect of the environment as there's no indication in the README that it wouldn't be expected to work. If you don't want people to use non-standalone, please either state clearly that there's no support, or don't mention it as a valid option.

jboehmler, 2018-06-24:
Maybe update the Makefile to similar to below?

ifneq ($(STANDALONE),y)
        @echo "Espressif ESP8266 SDK is installed. Toolchain contains only Open Source components."
        @echo "To link external proprietary libraries add the following to each project's"
        @echo "Makefile immediately after the line containing LDFLAGS = -Teagle.app.v6.ld
:"
        @echo "    CFLAGS += -I$(TOP)/sdk/include"
        @echo "    LDFLAGS += -L$(TOP)/sdk/lib -L$(TOP)/sdk/ld"