pfalcon/esp-open-sdk #288: Build on OSX fails with a

DouglasPearless, 2017-08-29:
Hi,

Followed the instructions and the make failed with:

```
[INFO ]  =================================================================
[INFO ]  Installing cross-gdb
[ERROR]    configure: error: python is missing or unusable
[ERROR]    make[4]: *** [configure-gdb] Error 1
[ERROR]    make[3]: *** [all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@216]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 35:19.00)
[35:19] / make[2]: *** [build] Error 1
```

Thoughts?

DouglasPearless, 2017-08-29:
There was no build.log file that I could find

xs23933, 2017-08-30:
[ERROR]    configure: error: python is missing or unusable

DouglasPearless, 2017-08-30:
python --version
Python 2.7.9
is what I have installed, using maCOS Sierra 10.12.6 (16G29)

SConaway, 2017-10-29:
Same here. Any luck?

mjcross, 2018-01-19:
I too encountered this problem, and in my case it was because I was missing the python-dev headers. I fixed it by reinstalling python with brew and doing `brew link --overwrite python`.

Juppit, 2018-01-19:
I got the same problem under Cygwin.
The fix was `apt-cyg install python2-devel`.

daviddpd, 2018-12-23:
Yes, dependency issues - I seemed to have figured it out on 10.14, not sure if this will fix 10.13 ... see my helper script on #342.