pfalcon/esp-open-sdk #291: Number of concurrent sockets limited to 5

rikte88, 2017-09-13:
Hi, as I understand, lwip limits the number of concurrent sockets to MEMP_NUM_TCP_PCB, which is defined in lwip/lwip/config/lwipopts.h as:
#define MEMP_NUM_TCP_PCB (*((volatile uint32*)0x600011FC))
Printing the u32_t at this memory address results in 5. Is there any way this could be increased? I tried changing it to:
#define MEMP_NUM_TCP_PCB 10
but it does not seem to work.
Best regards
Karl

nochkin, 2017-09-14:
I believe it's "MEMP_NUM_NETCONN".
I add "-DMEMP_NUM_NETCONN=8" when building my stuff and need more connections.

rikte88, 2017-09-14:
Thanks, I will try later today! I accidentally pasted without a code block, that address should be dereferenced, edited. However, I believe both need to be increased for TCP sockets. At https://github.com/mziwisky/esp8266-dev/blob/master/esphttpd/include/lwipopts.h it is defined as simply 5 with the comment MEMP_NUM_TCP_PCB: the number of simulatenously active TCP connections.

nochkin, 2017-09-15:
Interestingly enough, I use "MEMP_NUM_NETCONN" to increase the number of connections specifically for esphttpd project.
To be completely honest, I build it using esp-open-sdk tools while under esp-open-rtos.