pfalcon/esp-open-sdk #292: crosstool-ng failed when making with `make STANDALONE=n`

Kraust, 2017-09-15:
```kraust@Kiseki:/mnt/c/Users/secre/esp-open-sdk$ make STANDALONE=n
Makefile:187: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
Makefile:178: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
cp -f 1000-mforce-l32.patch crosstool-NG/local-patches/gcc/4.8.5/
make -C crosstool-NG -f ../Makefile _toolchain
make[1]: Entering directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
../Makefile:187: warning: overriding recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../Makefile:178: warning: ignoring old recipe for target 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
./ct-ng xtensa-lx106-elf
make[2]: Entering directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Max Filippov <jcmvbkbc@gmail.com>
URL: http://www.esp8266.com/viewtopic.php?f=9&t=224

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/mnt/c/Users/secre/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20170914.202619
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[02:42] -
[INFO ]  Retrieving needed toolchain components' tarballs: done in 248.84s (at 04:15)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[10:00] \
[INFO ]  Extracting and patching toolchain components: done in 610.91s (at 14:26)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 62.78s (at 15:29)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 102.46s (at 17:11)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 83.15s (at 18:34)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[ERROR]    make[3]: *** [/mnt/c/Users/secre/esp-open-sdk/crosstool-NG/.build/src/isl-0.14/Makefile.in] Error 1
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing ISL for host'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_isl_backend[scripts/build/companion_libs/121-isl.sh@125]
[ERROR]  >>        called from: do_isl_for_host[scripts/build/companion_libs/121-isl.sh@73]
[ERROR]  >>        called from: do_companion_libs_for_host[scripts/build/companion_libs.sh@36]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 18:58.42)
[18:59] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
../Makefile:132: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/mnt/c/Users/secre/esp-open-sdk/crosstool-NG'
Makefile:128: recipe for target '/mnt/c/Users/secre/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/mnt/c/Users/secre/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2```

This is on Linux for Windows 10, I will try on a VM and see if I can get a better response.

The build.log for crosstool-NG is also included. 

For reference, I have automake-1.15 installed:

```kraust@Kiseki:/mnt/c/Users/secre/esp-open-sdk$ automake --version
automake (GNU automake) 1.15
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

Written by Tom Tromey <tromey@redhat.com>
       and Alexandre Duret-Lutz <adl@gnu.org>.```

[build.txt](https://github.com/pfalcon/esp-open-sdk/files/1304977/build.txt)

maximkulkin, 2017-10-25:
Make sure you have enough space on device you're building on. Building toolchain requires A LOT of space (instructions recommend you have at least 10Gig of disk space). I was getting mysterious errors when I have created a smaller disk (I'm building on Mac OS and had to create a case-sensitive volume per instructions in README).

bureaua, 2017-10-29:
Same error. 
In my case I have ~250Gb free space

grodansparadis, 2017-10-29:
Download autmake 1.14 from [here](http://ftp.gnu.org/gnu/automake/) and put the install folder in you path and continue the build might help.

maximkulkin, 2017-10-30:
@grodamsparadis Automake version is not a problem, I was able to successfully build SDK in Ubuntu 16.04 with Automake 1.15.

Just in case you guys won't be able to build it on your system, here is Dockerfile to build an image with it:
```Dockerfile
FROM ubuntu:16.04 as builder

RUN groupadd -g 1000 docker && useradd docker -u 1000 -g 1000 -s /bin/bash --no-create-home
RUN mkdir /build && chown docker:docker /build

RUN apt-get update && apt-get install -y \
  make unrar-free autoconf automake libtool gcc g++ gperf \
  flex bison texinfo gawk ncurses-dev libexpat-dev python-dev python python-serial \
  sed git unzip bash help2man wget bzip2 libtool-bin

RUN su docker -c " \
    git clone --recursive https://github.com/pfalcon/esp-open-sdk.git /build/esp-open-sdk ; \
    cd /build/esp-open-sdk ; \
    make STANDALONE=n ; \
"

FROM ubuntu:16.04
RUN apt-get update && apt-get install -y make python python-serial
COPY --from=builder /build/esp-open-sdk/xtensa-lx106-elf /opt/xtensa-lx106-elf
ENV PATH /opt/xtensa-lx106-elf/bin:$PATH
```