pfalcon/esp-open-sdk #294: Crosstool-NG error - Can't solve it

carlosdomenje, 2017-11-02:
i have a problem with esp-open-sdk compilation. When i try to compile the IoT_Demo example from ESP8266_NONOS_SDK_V2.0.0_16_08_10 folder, I have this error.

`../../Makefile:105: atención: se anulan las instrucciones para el objetivo 'clean'
../Makefile:301: atención: se ignoran las instrucciones viejas para el objetivo 'clean'
../../Makefile:187: atención: se anulan las instrucciones para el objetivo 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
../../Makefile:178: atención: se ignoran las instrucciones viejas para el objetivo 'ESP8266_NONOS_SDK_V2.0.0_16_08_10/.dir'
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: *** crosstool-NG: No existe el archivo o el directorio.  Alto.
../../Makefile:142: fallo en las instrucciones para el objetivo 'crosstool-NG/ct-ng'
make: *** [crosstool-NG/ct-ng] Error 2`

I don't know what is happening, could you help me?

pfalcon, 2017-11-02:
Я не могу, я не понимаю языка, на котором написаны ваши сообщения об ошибках.

carlosdomenje, 2017-11-02:
'../../Makefile:105: Attention: the instructions for the 'clean' objective are canceled
../Makefile:301: attention: the old instructions for the 'clean' objective are ignored
../../Makefile:187: Attention: the instructions for the objective 'ESP8266_NONOS_SDK_V2.0.0_16_08_10 / .dir' are canceled
../../Makefile:178: Attention: old instructions are ignored for the purpose 'ESP8266_NONOS_SDK_V2.0.0_16_08_10 / .dir'
You cloned without --recursive, fetching submodules for you.
git submodule update --init --recursive
make -C crosstool-NG -f ../Makefile _ct-ng
make [1]: *** crosstool-NG: The file or directory does not exist. High.
../../Makefile:142: failure in the instructions for the target 'crosstool-NG / ct-ng'
make: *** [crosstool-NG / ct-ng] Error 2'

English version

pfalcon, 2017-11-02:
This seems to be the main error:

~~~
make -C crosstool-NG -f ../Makefile _ct-ng
make [1]: *** crosstool-NG: The file or directory does not exist. High.
~~~

You have your system before you - you can see if what it says is true. That doesn't make much sense, because "crosstool-NG" directory is the part of this repository, so it should be there just after the git clone, and definitely after `git submodule update --init --recursive`.

You may also want to specify what distro you use and `make --version`, so other users of those could help.

carlosdomenje, 2017-11-02:
My Linux distro is Ubuntu 16.04.3 LTS 64bits

`make --version
GNU Make 4.1
This program was built for x86_64-pc-linux-gnu
`

sifourquier, 2019-01-13:
look at https://github.com/pfalcon/esp-open-sdk/issues/340