pfalcon/esp-open-sdk #296: ANN: Key reinstallation attack (KRACK) situation

pfalcon, 2017-11-18:
This repository has added support for upstream vendor SDK which is specified by Espressif as having fixes  for KRACK vulnerability, in https://github.com/pfalcon/esp-open-sdk/commit/11a7e7e57f11b3685adf406775de8d1aa556ff49 commit. Note, that at the time of the writing, there's no official release of the SDK with the vulnerability fix from Espressif (only untagged commit in their binary blobs git repository).

Per this project policy, it's not immediately made the default vendor SDK used, until new version is fully validated for backwards compatibility and regression with a well known working version of the vendor SDK (2.0.0 currently).

To build the new version:

~~~
make clean
# Or, if you know what you're doing:
# make clean-sdk
# make clean-sysroot
make VENDOR_SDK=2.1.0-18-g61248df
~~~

Interested parties are welcome to validate the new SDK for backwards compatibility and regressions, and share experiences here. Note that the default validation testcase for esp-open-sdk is MicroPython ESP8266 port, as explained in the README. So, ideal testing would involve building MicroPython with both vendor SDK 2.0.0 and 2.1.0-18-g61248df and comparing MicroPython behavior over wide range of applications (using networking functionality first of all).

fake-fur, 2018-01-16:
this seems rather important to move along - what can we as users of this project do to help exactly? thanks

pfalcon, 2018-02-08:
Ok, so this definitely going overdue, so I'm just switching the default version to 2.1.0-18-g61248df, and we can go from there.

uzi18, 2018-02-09:
SDK 2.2.0 is available