pfalcon/esp-open-sdk #298: failed to download binutils-2.25.1.tar.xz

wemos, 2017-12-28:
Failed to build the crosstool-NG,

[EXTRA]    Extracting 'binutils-2.25.1'
[DEBUG]    ==> Executing: 'mkdir' '-p' 'binutils-2.25.1' 
[DEBUG]    ==> Executing: 'tar' '--strip-components=1' '-C' 'binutils-2.25.1' '-xv' '-f' '-' 
[FILE ]    tar: This does not look like a tar archive
[FILE ]    tar: Skipping to next header
[FILE ]    tar: Exiting with failure status due to previous errors
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Extract[scripts/functions@982]
[ERROR]  >>        called from: do_binutils_extract[scripts/build/binutils/binutils.sh@38]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@615]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 15:46.22)

I find in https://ftp.gnu.org/gnu/binutils 
there is not file named binutils-2.25.1.tar.xz
Where can I download the "binutils-2.25.1.tar.xz"?

dana321, 2018-06-06:
In your config file, change CT_CONFIGURE_has_xz to CT_CONFIGURE_has_xz=n

papadeltasierra, 2018-09-28:
Which config file? Please be very specific for those of use who only know enough to be dangerous ;-).

papadeltasierra, 2018-09-28:
Can you explain how your suggested fix helps?  Are you familiar with this:
[Problems with #esp-open-sdk build on Debian jessie #ESP8266 solved](http://grodansparadis.com/wordpress/?p=2802)?  I seem to be seeing the same issue in that the binutils...xz file is not being downloaded (I get an HTML error report downloaded into the file thus the error indicated in the build.log).  I'll see if I can figure out what's up but I suspect the binutils...xz file has perhaps moved to some other page on the web?

dana321, 2018-09-28:
The issue relates to crosstool-ng. In your crosstool configuration file, modify or add the parameter.

papadeltasierra, 2018-09-28:
dana321 - specifically which file?   I can see a number of config files that are autogenerated but a grep through the entire codebase does not seem to reveal a source file with this config parameter in it.  However this is merely a poor workaround for a bug.  The error occurs because the code tries to pull files from http://ftp.kernel.org and this fails.  Now many ISPs intercept the 'that website does not exist' response and produce their own web-page to return to users.  So in my case your scripts save off a file binutils...tax.xz that contains an HTML webpage generated by TalkTalk, my ISP.  The fix is to correct code that ends up in the 'functions' file to check the actually file that it thinks is downloaded - however I can't get this to work and cannot see why!  The code I'm trying to make work is this (around line 731):
`
            if [ -f "${CT_TARBALLS_DIR}/${file}${ext}" ]; then
                IS_HTML=$(file --brief "${CT_TARBALLS_DIR}/${file}${ext}" | grep HTML)
                if [ -z $IS_HTML ]; then
                    CT_DoLog DEBUG "Got '${file}' from the Internet"
                    CT_SaveLocal "${CT_TARBALLS_DIR}/${file}${ext}"
                    return 0
                fi
            fi
`
but the build system complains about the line that runs 'file' (IS_HTML=...).  Since the output of the error appears to be swallowed somewhere, I have no idea why and all attempts to rewrite this so that it works have failed.  Are you able to either shed any light on why or tell me how to actually see the error that the build system complains about?

dana321, 2018-09-29:
I have talktalk as well.  Make sure when you cloned this repository, you added the --recurse-submodules in the git clone command, its in the crosstools-NG subproject that this option exists.

Btw, this is NOT my project, i only found this error with another project relating to crosstools-NG, changing the source code in one place won't fix the problem, there is multiple places.  The mirror doesn't contain an xz file, so you need to use .gz (so skip xz)

scripts/functions contains the environment option:
`
# Build up the list of allowed tarball extensions
# Add them in the prefered order; most preferred comes first
CT_DoListTarballExt() {
    if [ "${CT_CONFIGURE_has_xz}" = "y" ]; then
        printf ".tar.xz\n"
    fi
    if [    "${CT_CONFIGURE_has_lzma}" = "y"    \
         -o "${CT_CONFIGURE_has_xz}" = "y" ]; then
        printf ".tar.lzma\n"
    fi
    printf ".tar.bz2\n"
    printf ".tar.gz\n.tgz\n"
    printf ".tar\n"
    printf ".zip\n"
}               
`

papadeltasierra, 2018-09-30:
Thanks for the info and help.  I've been trying to make sense of this because we you say the same code seems to get put all over the place but I'm going to get my head out of the details and try and get this installed.  thanks again.

closed 2022-09-01 by wemos