pfalcon/esp-open-sdk #305: make error: esp-open-sdk/crosstool-NG': Permission denied

youngda, 2018-02-11:
### First of all

：

Now configured for "xtensa-lx106-elf"
gmake[2]: Leaving directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/root/wifiesp/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
gmake[2]: Entering directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
[ERROR]  You must NOT be root to run crosstool-NG
[00:00] / gmake[2]: *** [build] Error 1
gmake[2]: Leaving directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
make: *** [/root/wifiesp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
[root@bemfa esp-open-sdk]# ls

### then I create a new user, change permissions

[youngda@bemfa esp-open-sdk]$ make
Makefile:184: warning: overriding recipe for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
Makefile:179: warning: ignoring old recipe for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
make: stat: /root/wifiesp/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc: Permission denied
make -C crosstool-NG -f ../Makefile _ct-ng
make[1]: Entering directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
../Makefile:184: warning: overriding recipe for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
../Makefile:179: warning: ignoring old recipe for target `ESP8266_NONOS_SDK-2.1.0-18-g61248df/.dir'
./bootstrap
Running autoconf...
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
fatal: Could not change back to '/root/wifiesp/esp-open-sdk/crosstool-NG': Permission denied
cannot stat initial working directory for /root/wifiesp/esp-open-sdk/crosstool-NG: Permission denied at /usr/share/autoconf/Autom4te/General.pm line 196.
make[1]: *** [_ct-ng] Error 1
make[1]: Leaving directory `/root/wifiesp/esp-open-sdk/crosstool-NG'
make: *** [crosstool-NG/ct-ng] Error 2

uzi18, 2018-02-11:
Start from beginning,  remove all filles and fetch all to user's home dir. 
Launch setup from there

youngda, 2018-02-19:
@uzi18 You're right, thank you.