pfalcon/esp-open-sdk #306: crosstool-ng build error - wrong link to MPC tarball ?

Koperlite, 2018-02-11:
Hello, 

I've had an error with the command : 
 ./ct-ng build

In fact the build process try to get MPC tarball from the 'directory' 'http://www.multiprecision.org/mpc/download/' but it is not in this place anymore (the directory is not existing). The new link to the tarball from multiprecision.org is : https://ftp.gnu.org/gnu/mpc/mpc-1.0.2.tar.gz

I got it and copied it in the .build/.../tarball directory. It worked arround.

Regards.

The error message is : 
```
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180211.112822
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_mpc_get[scripts/build/companion_libs/140-mpc.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@583]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 0:02.20)
[00:02] / ct-ng:148: recipe for target 'build' failed
make: *** [build] Error 1
```

Some interesting lines from build.log
```
[EXTRA]     Retrieving 'mpc-1.0.2'
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.gz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.tar.gz.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.gz' 
[ALL  ]    --2018-02-11 11:28:23--  http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.gz
[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.gz"
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.xz'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.tar.xz.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.xz' 
[ALL  ]    --2018-02-11 11:28:23--  http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.xz
[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.xz"
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.lzma'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.tar.lzma.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.lzma' 
[ALL  ]    --2018-02-11 11:28:23--  http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar.lzma

```

pfalcon, 2018-02-11:
Please see https://github.com/pfalcon/esp-open-sdk/issues/100

romavita, 2018-11-26:
Your fix was just what I needed.

jeancode, 2019-01-18:
erics465  apenas pruebo el parche veremos si eres verga.

qiuwenhui, 2019-04-25:
Downloads the specified utility zip to the directory,successful