pfalcon/esp-open-sdk #307: Issues with gunzip expat-2.1.0.tar.gz

xianrecords2015, 2018-02-14:
***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-lx106-elf"
make[2]: Leaving directory '/opt/esp-open-sdk/crosstool-NG'
sed -r -i.org s%CT_PREFIX_DIR=.*%CT_PREFIX_DIR="/opt/esp-open-sdk/xtensa-lx106-elf"% .config
sed -r -i s%CT_INSTALL_DIR_RO=y%"#"CT_INSTALL_DIR_RO=y% .config
cat ../crosstool-config-overrides >> .config
./ct-ng build
make[2]: Entering directory '/opt/esp-open-sdk/crosstool-NG'
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180213.225313
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 117.62s (at 01:59)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[02:00] / 
gzip: /opt/esp-open-sdk/crosstool-NG/.build/tarballs/expat-2.1.0.tar.gz: not in gzip format
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Extract[scripts/functions@985]
[ERROR]  >>        called from: do_expat_extract[scripts/build/companion_libs/210-expat.sh@17]
[ERROR]  >>        called from: do_companion_libs_extract[scripts/build/companion_libs.sh@22]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@614]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 1:59.48)
[02:00] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 2
make[2]: Leaving directory '/opt/esp-open-sdk/crosstool-NG'
../Makefile:133: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/opt/esp-open-sdk/crosstool-NG'
Makefile:129: recipe for target '/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/opt/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
parallels@parallels-vm:/opt/esp-open-sdk$ cd /opt/esp-open-sdk/crosstool-NG/.build/tarballs/

omarkmsft, 2018-02-24:
its failing to download expat-2.1.0.tar.gz from the web.  Search for that file on the web and manually download it into the opt/esp-open-sdk/crosstool-NG/.build/tarballs/ replacing the file that is already there.

martinjirku, 2018-02-25:
It seems that sourceforge is not reliable source for this repository. It failed to download it for me as well. 
Expat library move from sourceforge to github, see expat/expat#156.

I am curious, why is used here the fork https://github.com/jcmvbkbc/crosstool-NG instead of original repository: https://github.com/jcmvbkbc/crosstool-NG