pfalcon/esp-open-sdk #308: Error installing cross-gdb

GTP95, 2018-02-17:
While making the standalone version on Netrunner (based on Arch Linux) I got the following error:

> [INFO ]  Installing cross-gdb
[ERROR]    /home/giacomo/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:231:47: error: dereferencing pointer to incomplete type 'scm_t_port {aka struct scm_t_port}'
[ERROR]    /home/giacomo/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:681:19: error: 'SCM_PORT_NEITHER' undeclared (first use in this function); did you mean 'SCM_PORT_TYPE'?
[ERROR]    /home/giacomo/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:711:19: error: 'SCM_PORT_NEITHER' undeclared (first use in this function); did you mean 'SCM_PORT_TYPE'?
[ERROR]    /home/giacomo/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:790:24: error: 'SCM_PORT_WRITE' undeclared (first use in this function); did you mean 'SCM_PORT_TYPE'?
[ERROR]    /home/giacomo/esp-open-sdk/crosstool-NG/.build/src/gdb-7.10/gdb/guile/scm-ports.c:816:29: error: 'SCM_PORT_READ' undeclared (first use in this function); did you mean 'SCM_PORT_WRITE'?
[ERROR]    make[5]: *** [Makefile:2469: scm-ports.o] Error 1
[ERROR]    make[4]: *** [Makefile:8755: all-gdb] Error 2
[ERROR]    make[3]: *** [Makefile:845: all] Error 2
[ERROR]   
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing cross-gdb'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_debug_gdb_build[scripts/build/debug/300-gdb.sh@120]
[ERROR]  >>        called from: do_debug[scripts/build/debug.sh@35]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]   
[ERROR]  (elapsed: 19:47.93)
[19:48] / make[2]: *** [ct-ng:152: build] Error 2
make[2]: uscita dalla directory "/home/giacomo/esp-open-sdk/crosstool-NG"
make[1]: *** [../Makefile:137: _toolchain] Error 2
make[1]: uscita dalla directory "/home/giacomo/esp-open-sdk/crosstool-NG"
make: *** [Makefile:130: /home/giacomo/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2
Here is a link to my build.log file in the crosstool-NG directory: https://app.box.com/s/w2zytu4hozn59p1902te78pfqtq0jhja
What can I try to successfully compile the SDK?

GTP95, 2018-02-18:
Nevermind, I successfully build it using an Ubuntu Vagrant image.

RenaKunisaki, 2019-11-10:
Same issue here.

shooter0106, 2020-12-27:
You may disable cross-gdb building by setting variable `CT_DEBUG_gdb` to `n` value in `crosstool-NG/samples/xtensa-lx106-elf/crosstool.config` config