pfalcon/esp-open-sdk #312: Issues with building esp-open-sdk

inventvictor, 2018-03-02:
**This build keeps failing and really its frustrating!!!

Some issues with not finding some tarball stuff..

Is there some issue with FTP proxy?**

```
[EXTRA]    Extracting 'expat-2.1.0'
[DEBUG]    ==> Executing: 'mkdir' '-p' 'expat-2.1.0'
[DEBUG]    ==> Executing: 'tar' '--strip-components=1' '-C' 'expat-2.1.0' '-xv' '-f' '-'
[FILE ]    tar: This does not look like a tar archive
[FILE ]    tar: Exiting with failure status due to previous errors
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: CT_Extract[scripts/functions@985]
[ERROR]  >>        called from: do_expat_extract[scripts/build/companion_libs/210-expat.sh@17]
[ERROR]  >>        called from: do_companion_libs_extract[scripts/build/companion_libs.sh@22]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@614]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6f/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 69:09.85)
```

DeuxVis, 2018-03-02:
Can you give some information about your environment ?
I built it yesterday night from a fresh github extract and it worked perfectly.

inventvictor, 2018-03-02:
Got this error on Ubuntu 16.04. Sometimes this same error shows up on Ubuntu 17.10

[ERROR]    The 'expat-2.1.0' sources were partially extracted.
[ERROR]    Please remove first:
[ERROR]     - the source dir for 'expat-2.1.0', in '/home/bruce/espn/esp-open-sdk/crosstool-NG/.build/src'
[ERROR]     - the file '/home/bruce/espn/esp-open-sdk/crosstool-NG/.build/src/.expat-2.1.0.extracting'
[ERROR]    I'll stop now to avoid any carnage...
[ERROR]
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Extracting and patching toolchain components'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_Abort[scripts/functions@331]
[ERROR]  >>        called from: CT_Extract[scripts/functions@959]
[ERROR]  >>        called from: do_expat_extract[scripts/build/companion_libs/210-expat.sh@17]
[ERROR]  >>        called from: do_companion_libs_extract[scripts/build/companion_libs.sh@22]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@614]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/crosstool-ng-1.22.0-60-g37b07f6/B - Known issues.txt'
[ERROR]
[ERROR]  (elapsed: 0:16.05)
[00:18] / ct-ng:152: recipe for target 'build' failed
make[2]: *** [build] Error 1
make[2]: Leaving directory '/home/bruce/espn/esp-open-sdk/crosstool-NG'
../Makefile:133: recipe for target '_toolchain' failed
make[1]: *** [_toolchain] Error 2
make[1]: Leaving directory '/home/bruce/espn/esp-open-sdk/crosstool-NG'
Makefile:129: recipe for target '/home/bruce/espn/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc' failed
make: *** [/home/bruce/espn/esp-open-sdk/xtensa-lx106-elf/bin/xtensa-lx106-elf-gcc] Error 2

DeuxVis, 2018-03-02:
I just retried, I now have the same problem, and this seems to be coming from sourceforge :

```
[INFO ]  Extracting and patching toolchain components
[12:33] / 
gzip: ~/SRC/esp-open-sdk/crosstool-NG/.build/tarballs/expat-2.1.0.tar.gz: not in gzip format
[ERROR]   
```

```
~/SRC/esp-open-sdk$ cat ~/SRC/esp-open-sdk/crosstool-NG/.build/tarballs/expat-2.1.0.tar.gz
<html><head>
<title>SourceForge</title>
</head><body>
<noscript>
We're sorry -- the Sourceforge site is currently in Disaster Recovery mode, and currently requires
the use of javascript to function.  Please check back later.
</noscript>
</body></html>
```

Maybe try to download the tarball using a browser, or wait till sourceforge are fixed.

DeuxVis, 2018-03-02:
I downloaded that tarball from sourceforge into the 'crosstool-NG/.build/tarballs/' folder, and did what your 2nd error message instructed (got it too) :
`~/SRC/esp-open-sdk$ rm -Rfv crosstool-NG/.build/src/expat-2.1.0 crosstool-NG/.build/src/.expat-2.1.0.extracting`

and now the build passed that step - it is still ongoing so maybe some other sourceforge issues will need the same kind of workaround later on.

DeuxVis, 2018-03-02:
By the way I use an Ubuntu 14.04 VM for testing at the moment.

DeuxVis, 2018-03-02:
Workaround verified to work on another ubuntu 14.04

Download for expat is at https://sourceforge.net/projects/expat/files/expat/2.1.0/

Maybe that issue can be closed ?

inventvictor, 2018-03-03:
Thanks 

This is great news

closed 2018-03-03 by inventvictor