pfalcon/esp-open-sdk #313: Compiling ESP_OPEN_SDK

GanesanGuru, 2018-03-16:
Hi fellas,

I just now started working with esp-open-sdk. All I've doubts are basic one's, yet they are vital. I cloned the git and I've ran the "make" command. It generated four files.
*sdk - soft link for ESP_SDK generated
*ESP_SDK itself
*xtensa-lx106-elf
*user_rf_cal_sector_set.o

What is xtensa-lx106-elf? Is it compiler or a toolchain? Because esp-open-sdk/xtensa-lx106-elf/bin has all in it (compiler,assembler,linker,debugger). SO HAS TO BE A **TOOLCHAIN**??

Similarly explain what are crosstool-NG,lx106-hal,esp-open-lwip which are downloaded from git hub.

PS:I'm newbie to these knid of things and all. So please suggest me what are the things that I should go through to understand these.

Rosmianto, 2018-03-30:
Basically we are compiling a compiler. We want to target ESP8266, so we use xtensa-lx106-elf.

We want to compile xtensa-lx106-elf by ourselves, so we use crosstool-NG to do that. Xtensa-lx106- is both a compiler and a toolchain.

GanesanGuru, 2018-04-02:
Hi Rosmianto,

Thanks for helping me out. Correct me if I'm wrong.

As a reference to ESP8266 hardware, we use the HAL of Tensilica L106  as **lx106-hal**, which is the target hardware. Toolchain is _combo of (Compiler+Linker+Libraries+Debugger) for a specific hardware_ and to create a toolchain for ESP8266, we use **crosstool-NG**, which is a cross-platform toolchain generator (as we create it in linux platform,which is different from target platform,which is Tensilica L106 ,Xtensa Architecture). 

- Compiler is gcc, which supports Xtensa architecture also. Its available as free as internet.

- Linker is GNU binutils which is also available as free as internet.

- Debugger is GNU Debugger, which is also available as free as internet.

- Library provides interfaces for OS_______________________ _I'm not sure about this one alone. What does it provide really? As device drivers,peripherals drivers,board support packages are provided by HAL itself. Hence actually what does it really provide?_

Hence all the package needed for toolchain creation are download and then harware specific toolcahin is builded from the scrape using these packages, building tools (GNU autoconf) and crosstool-NG. HENCE TOOLCHAIN IS CREATED.

What is SDK then? The aim of esp-open-sdk is creating toolchain and sdk for ESP8266. Toolchain is part is understood. They tell sdk is like IDE, but without that much comfortness. IDE is GUI enabled sdk, for program writing,compiling,debugging & downloading code into hardware. So SDK also talks about writing,compiling,linking,debugging,downloading.....which is TOOLCHAIN components. So please differentiate between toolchain and SDK.

GanesanGuru, 2018-04-06:
Hi guys,

I came across another info about bootloaders in ESP. Help me out  if I'm wrong.

1.ESP resetted........... Gets the reset vector address from Vector table as 0x40000080, which is BOOT-ROM
2.Program in Boot-ROM is pre-loaded. Which takes care of initialization of hardware and loading/booting of (code+secondary bootloader) to Flash memory via SPI................................ based on specific GPIO pin's state alone
3. Then move the code+secondary bootloader to RAM and handover the control to the secondary bootloader.
4.Now secondary bootloader takes care of execting the application code.

**The doubt is,  where is the secondary bootloader..... in readme doc, for BOOT/OTA firmware, its boot.bin. But no such file is there for Non-BOOT/Non-OTA firmware!!!**