pfalcon/esp-open-sdk #316: Cygwin build is non-functional

MayaPosch, 2018-04-04:
According to the project README, it should be possible to build the SDK on Windows using Cygwin. After spending half the day on trying to resolve various dependencies and issues, I had to give up on trying to make it work. 

Unless a guide can be added for building the SDK under Windows, the mention of Cygwin should be removed from the project as it's clearly not a functional port.

Juppit, 2018-04-08:
Try to add the following lines to crosstool-config-overrides:
```
# to prevent segfault on Cygwin
CT_CC_GCC_DISABLE_PCH=y
# for GDB on Cygwin64
CT_GDB_CROSS_EXTRA_CONFIG_ARRAY=--disable-tui
```
And  then try it with Cygwin again.

MayaPosch, 2018-04-15:
Thank you, that actually allows me to build the toolchain. After this I run into another error, though:

`configure:6132: result: no
configure:6149: checking ncurses/ncurses.h usability
configure:6149: gcc -c -g -O2  conftest.c >&5
configure:6149: $? = 0
configure:6149: result: yes
configure:6149: checking ncurses/ncurses.h presence
configure:6149: gcc -E  conftest.c
configure:6149: $? = 0
configure:6149: result: yes
configure:6149: checking for ncurses/ncurses.h
configure:6149: result: yes
configure:6162: checking for library containing initscr
configure:6193: gcc -o conftest.exe -g -O2   conftest.c  >&5
/tmp/ccYtmrGG.o: In function `main':
/home/Maya/esp-open-sdk/crosstool-NG/conftest.c:37: undefined reference to `initscr'
/home/Maya/esp-open-sdk/crosstool-NG/conftest.c:37:(.text.startup+0xa): relocation truncated to fit: R_X86_64_PC32 against undefined symbol `initscr'
collect2: error: ld returned 1 exit status
configure:6193: $? = 1
configure: failed program was:
| /* confdefs.h */
| #define PACKAGE_NAME "crosstool-NG"
| #define PACKAGE_TARNAME "crosstool-ng"
| #define PACKAGE_VERSION "crosstool-ng-1.22.0-60-g37b07f6f"
| #define PACKAGE_STRING "crosstool-NG crosstool-ng-1.22.0-60-g37b07f6f"
| #define PACKAGE_BUGREPORT "crossgcc@sourceware.org"
| #define PACKAGE_URL ""
| #define STDC_HEADERS 1
| #define HAVE_SYS_TYPES_H 1
| #define HAVE_SYS_STAT_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_STRING_H 1
| #define HAVE_MEMORY_H 1
| #define HAVE_STRINGS_H 1
| #define HAVE_INTTYPES_H 1
| #define HAVE_STDINT_H 1
| #define HAVE_UNISTD_H 1
| #define HAVE_STDLIB_H 1
| #define HAVE_MALLOC 1
| #define HAVE_STDLIB_H 1
| #define HAVE_REALLOC 1
| #define HAVE_ALLOCA_H 1
| #define HAVE_ALLOCA 1
| #define HAVE_NCURSES_NCURSES_H 1
| /* end confdefs.h.  */
|
| /* Override any GCC internal prototype to avoid an error.
|    Use char because int might match the return type of a GCC
|    builtin and then its argument prototype would still apply.  */
| #ifdef __cplusplus
| extern "C"
| #endif
| char initscr ();
| int
| main ()
| {
| return initscr ();
|   ;
|   return 0;
| }
configure:6193: gcc -o conftest.exe -g -O2   conftest.c -lncursesw   >&5
configure:6193: $? = 0
configure:6210: result: -lncursesw
configure:6221: checking for library containing tgetent
configure:6252: gcc -o conftest.exe -g -O2   conftest.c -lncursesw  >&5
configure:6252: $? = 0
configure:6269: result: none required
configure:6481: creating ./config.status
`

It's possible that some dependency is missing, as they were not listed for Cygwin.

Juppit, 2018-04-19:
Maybe:
apt-cyg install ncurses libncurses-devel

MayaPosch, 2018-04-19:
Both ncurses and libncurses-devel are installed.