pfalcon/esp-open-sdk #319: Retrieving needed toolchain components' tarball failed (again?)

ficolas2, 2018-04-07:
https://github.com/pfalcon/esp-open-sdk/issues/237
Im trying to build this following 
https://github.com/esp8266/esp8266-wiki/wiki/Toolchain
but im getting an error like the one in the issue linked above:

`[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2.tgz"
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/Espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.tar.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar' 
[ALL  ]    --2018-04-07 03:37:26--  http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar
[ALL  ]    Resolving www.multiprecision.org (www.multiprecision.org)... 185.233.100.48
[ALL  ]    Connecting to www.multiprecision.org (www.multiprecision.org)|185.233.100.48|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 500 Internal Server Error
[ALL  ]    2018-04-07 03:37:26 ERROR 500: Internal Server Error.
[ALL  ]    
[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2.tar"
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.zip'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/Espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.zip.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2.zip' 
[ALL  ]    --2018-04-07 03:37:26--  http://www.multiprecision.org/mpc/download/mpc-1.0.2.zip
[ALL  ]    Resolving www.multiprecision.org (www.multiprecision.org)... 185.233.100.48
[ALL  ]    Connecting to www.multiprecision.org (www.multiprecision.org)|185.233.100.48|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 500 Internal Server Error
[ALL  ]    2018-04-07 03:37:26 ERROR 500: Internal Server Error.
[ALL  ]    
[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2.zip"
[DEBUG]    Trying 'http://www.multiprecision.org/mpc/download/mpc-1.0.2'
[DEBUG]    ==> Executing: 'wget' '--passive-ftp' '--tries=3' '-nc' '--progress=dot:binary' '-T' '10' '-O' '/opt/Espressif/crosstool-NG/.build/tarballs/mpc-1.0.2.tmp-dl' 'http://www.multiprecision.org/mpc/download/mpc-1.0.2' 
[ALL  ]    --2018-04-07 03:37:26--  http://www.multiprecision.org/mpc/download/mpc-1.0.2
[ALL  ]    Resolving www.multiprecision.org (www.multiprecision.org)... 185.233.100.48
[ALL  ]    Connecting to www.multiprecision.org (www.multiprecision.org)|185.233.100.48|:80... connected.
[ALL  ]    HTTP request sent, awaiting response... 500 Internal Server Error
[ALL  ]    2018-04-07 03:37:26 ERROR 500: Internal Server Error.
[ALL  ]    
[DEBUG]    Not at this location: "http://www.multiprecision.org/mpc/download/mpc-1.0.2"
[ERROR]  
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Retrieving needed toolchain components' tarballs'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: do_mpc_get[scripts/build/companion_libs/140-mpc.sh@741]
[ERROR]  >>        called from: do_companion_libs_get[scripts/build/companion_libs.sh@15]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@583]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'share/doc/crosstool-ng/ct-ng.1.20.0/B - Known issues.txt'
[ERROR]  
[ERROR]  (elapsed: 2:04.36)`

ficolas2, 2018-04-07:
I was able to solve the issue by changing the url it was looking for the file in to a different one, that I found in http://www.multiprecision.org/mpc/download.html

TimotF, 2018-06-23:
I had the same issue and was able to fix it by downloading the archive v1.02 from your link and placing it in crosstool-NG/.build/tarballs

Thanks !

pfalcon, 2018-07-10:
Also #100.

closed 2018-07-10 by pfalcon

lomassubedi, 2018-11-13:
Solved as follows : 

- Downloaded tarbel _mpc-x.x.x.tar.gz_ from the link shared by @ficolas2 
- Copy it to _crosstool-NG/.build/tarballs/_
- `make`
- ` make install`
- .`/ct-ng build`