pfalcon/esp-open-sdk #323: IGMP - Timer bug in igmp_start_timer()

Cicero-MF, 2018-05-01:
Hi, I've found this exact problem:
https://github.com/esp8266/Arduino/issues/3498

UDP multicast messages eventually stop getting through.

I'm trying to test the suggested fix, but for some reason if I change the igmp.h file to reference os_random() instead, recompile the SDK, then compile my project, my project binaries dont change....so in essence the change isn't being carried through from esp-open-lwip to my project.

I know I'm probably doing something stupid/wrong, but I'm not sure how to compile to test the change.

Cicero-MF, 2018-06-06:
Is anyone supporting this thing anymore?

martin-ger, 2018-06-06:
Check, whether ./esp-open-sdk/xtensa-lx106-elf/xtensa-lx106-elf/sysroot/usr/lib/liblwip_open.a is the same as ./esp-open-sdk/esp-open-lwip/liblwip_open.a . The linker uses the first by default, the latter is build by make.

Cicero-MF, 2018-06-07:
@martin-ger Thanks for responding, I seriously appreciate it.  
They are the same....but I actually don't think I'm using the library in the first place - which points to a lack of understanding on my side of the entire open-sdk structure.  
All my wifi connections are done with espconn, which I presume still references the SDK blobs?  If I want to use esp-open-lwip,  do I have to replace my espconn functionality with lwip?  And also include lwip-open in my makefile instead of just lwip?

martin-ger, 2018-06-07:
Yes, the Makefile has to link lwip_open. If you need the esp_conn interface, you can compile and use the one from the lwip/app directory.

Cicero-MF, 2018-06-08:
Ok, I'll give it a go and report back here.  Thanks.  
Also given your fork of this repo a follow...you seem to be doing good further work to it.

Cicero-MF, 2018-06-13:
Changing the LWIP_RAND to os_random() didn't help in my case.  Multicast receive still cuts out after 4m40s.

Cicero-MF, 2019-01-04:
Oops, sorry forgot to follow up on this.  My issue although showing the same symptoms, was not a bug in regards to the open-sdk, but a secondary router downstream blocking after that timeframe.

closed 2019-01-04 by Cicero-MF